Xinyuan-LilyGO/TTGO_TWatch_Library #3: Compilation Errors

PhilColbert, 2019-11-17:
Trying to compile all this code runs into many errors, does anyone have a version that compiles?

thanks

PhilColbert, 2019-11-18:
Compiles on arduino but not as a component for espidf.... probably compilation flags :)

incyi, 2019-11-21:
Please tell me when you can compile it for esp-idf.

salem83, 2020-05-20:
@PhilColbert How did you manage to get it compiled?
i just installed Arduino 1.8.12 and ESP 1.0.4

which gives me the following errors:

`
In file included from C:\Users\Administrator\Documents\Arduino\libraries\TTGO_TWatch_Library\src\MPR121.h:5:0,

                 from C:\Users\Administrator\Documents\Arduino\libraries\TTGO_TWatch_Library\src\MPR121.cpp:29:

C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire\src/Wire.h: In member function 'uint8_t MPR121_Class::readRegister8(uint8_t)':

C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire\src/Wire.h:99:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int)

     uint8_t requestFrom(int address, int size);

             ^

C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire\src/Wire.h:97:13: note: candidate 2: uint8_t TwoWire::requestFrom(uint8_t, uint8_t)

     uint8_t requestFrom(uint8_t address, uint8_t size);

             ^

C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire\src/Wire.h: In member function 'uint16_t MPR121_Class::readRegister16(uint8_t)':

C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire\src/Wire.h:99:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int)

     uint8_t requestFrom(int address, int size);

             ^

C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire\src/Wire.h:97:13: note: candidate 2: uint8_t TwoWire::requestFrom(uint8_t, uint8_t)

     uint8_t requestFrom(uint8_t address, uint8_t size);

             ^

In file included from C:\Users\Administrator\Documents\Arduino\libraries\TTGO_TWatch_Library\src\BBQ10Keyboard.h:3:0,

                 from C:\Users\Administrator\Documents\Arduino\libraries\TTGO_TWatch_Library\src\BBQ10Keyboard.cpp:5:

C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire\src/Wire.h: In member function 'uint8_t BBQ10Keyboard::readRegister8(uint8_t) const':

C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire\src/Wire.h:99:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int)

     uint8_t requestFrom(int address, int size);

             ^

C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire\src/Wire.h:97:13: note: candidate 2: uint8_t TwoWire::requestFrom(uint8_t, uint8_t)

     uint8_t requestFrom(uint8_t address, uint8_t size);

             ^

C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire\src/Wire.h: In member function 'uint16_t BBQ10Keyboard::readRegister16(uint8_t) const':

C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire\src/Wire.h:99:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int)

     uint8_t requestFrom(int address, int size);

             ^

C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire\src/Wire.h:97:13: note: candidate 2: uint8_t TwoWire::requestFrom(uint8_t, uint8_t)

     uint8_t requestFrom(uint8_t address, uint8_t size);

             ^

[... 49 lines cut ...]

libraries\TTGO_TWatch_Library\TFT_eSPI\Fonts\Font72x53rle.c.o:(.rodata.chr_f72_38+0x0): multiple definition of `chr_f72_38'

libraries\TTGO_TWatch_Library\TFT_eSPI\Fonts\Font72rle.c.o:(.rodata.chr_f72_38+0x0): first defined here

libraries\TTGO_TWatch_Library\TFT_eSPI\Fonts\Font72x53rle.c.o:(.rodata.chr_f72_39+0x0): multiple definition of `chr_f72_39'

libraries\TTGO_TWatch_Library\TFT_eSPI\Fonts\Font72rle.c.o:(.rodata.chr_f72_39+0x0): first defined here

libraries\TTGO_TWatch_Library\TFT_eSPI\Fonts\Font72x53rle.c.o:(.rodata.chr_f72_3A+0x0): multiple definition of `chr_f72_3A'

libraries\TTGO_TWatch_Library\TFT_eSPI\Fonts\Font72rle.c.o:(.rodata.chr_f72_3A+0x0): first defined here

libraries\TTGO_TWatch_Library\TFT_eSPI\Fonts\Font72x53rle.c.o:(.rodata.widtbl_f72+0x0): multiple definition of `widtbl_f72'

libraries\TTGO_TWatch_Library\TFT_eSPI\Fonts\Font72rle.c.o:(.rodata.widtbl_f72+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "SD.h"
 Used: C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
exit status 1
Error compiling for board TTGO T-Watch.
`

lewisxhe, 2020-05-21:
This is my mistake, I upgraded TFTeSPI, I only tested in platformIO, it will indeed cause an error in Arduino, I will roll back the code first, you can pull it again @salem83

salem83, 2020-05-24:
@lewisxhe thanks. unfortunately, I still got the same errors in the Arduino IDE. 
But i preferrably work in PlatformIO where i ahve no problems.

closed 2020-06-28 by lewisxhe

Fusseldieb, 2021-01-22:
Hijacking this thread to post my solution about the compilation errors I had and struggled for days!
My error was that something was wrong with the Ticker library, but yours could differ.

The solution is, at least if you're using PlatformIO, to **uninstall every library available in the "Global Storage"**, mine had a library there called Ticker, which WASN'T the same as this library used, and thus gave errors.

[image]

Then install the libraries needed for your Project one by one and be sure to select your active TTGO project when installing those! See below:
[image]

And now try to compile again. Should now work!