Xinyuan-LilyGO/TTGO_TWatch_Library #4: USB-C to USB-C not working

incyi, 2020-02-03:
When I connect my T-Watch with an USB-C to USB-C cable on my MacBook Pro 2018, it does not work. But when I connect an USB-C to USB-A adapter in my MacBook. And use an USB-A to USB-C cable, it works. I already tried different USB-C cables.

Is this a hardware or software issue ?

lewisxhe, 2020-02-04:
If there is a hardware problem, then you should not have A-> C, so I think this is a Mac problem! I use macbook air A-> C and it works fine

incyi, 2020-02-04:
I'll try another laptop with USB-C.

incyi, 2020-05-07:
Apple is stupid, it’s really only on the mac not working.

incyi, 2020-05-25:
Oke, this is getting really strange. I found some cables, which get it working on the mac. I think this is an USB-C cable issue.

closed 2020-05-25 by incyi