Xinyuan-LilyGO/TTGO_TWatch_Library #5: Using two I2C buses

glfurtado61, 2020-03-29:
Hi. I'm using GPIO25 and GPIO26 (on expansion connector) to comunicate by I2C. I have suceeded but touching the screen was out. I'm using Wire1 on extension communication, on IDE Arduino. What can I do to solve this issue?

lewisxhe, 2020-03-29:
Wire1 is already used in touch screen, if you need to use 25,26, then you can only use softI2C

glfurtado61, 2020-03-29:
OK, Lewis He. Thank you for your reply.
Can you suggest a library that´s ok for use with T-Watch? Some in special?
Thanks in advance
Gustavo

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Livre
de vírus. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>.
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

lewisxhe, 2020-03-29:
You can use https://github.com/pasko-zh/brzo_i2c

glfurtado61, 2020-03-29:
OK, thanks. I'll try that!

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Livre
de vírus. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>.
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2020-06-28 by lewisxhe