Xinyuan-LilyGO/TTGO_TWatch_Library #6: Lora sample don't work

georaura, 2020-04-03:
HI, 
I am trying to compile the lora example using the arduino-LoRa library, the program compiles without errors, but stops when initializing the lora module.
Specifically it stops at the following code block:

--------------
int ret = 0; 
do {
        lv_task_handler();
        ret = LoRa.begin(BAND);
        delay(500);
    } while (!ret);
----------------
Any suggestion?

lewisxhe, 2020-04-07:
If you run the sample program and cannot initialize normally, the first step is to exclude whether the FPC cable connection is normal, and then consider whether there is a hardware problem with the lora board.

lewisxhe, 2020-04-07:
You also need to determine whether the frequency accepted by the module is used, defined in line 24 in lora.ino

georaura, 2020-04-10:
I define the frequency in line 24 in lora.ino at 915 (#define LORA PERIOD 915), by the result is the same: the program compiles without errors, but stops with "**Begin Lora Module**" message.  
With the st7xg lora example I have not any problem, then I think Isn't a hardware problem with the lora board.

lewisxhe, 2020-04-11:
Oh, are you using S76G? Then you can't run this /examples/lora/lora.ino, this is for another board

georaura, 2020-04-17:
Yes, I'm using S76G. Then isn't possible use arduino-LoRa library  with this board?

lewisxhe, 2020-05-09:
S76G has built-in firmware, so it can't be used.

closed 2020-06-28 by lewisxhe