Xinyuan-LilyGO/TTGO_TWatch_Library #8: Unable to establish GPS position with S7xG

jsonpoindexter, 2020-05-08:
Hello,

I have a TTGO ESP32 watch with the S7XG LORA/GPS board. I am current unable to obtain a GPS position. It has been running in GPS mode and repeatedly displaying `>>POSITIONING (Xs)`
It has been running for 2+ hours and still has not acquired a GPS position. Please advise.

Code I am running:
https://raw.githubusercontent.com/Xinyuan-LilyGO/TTGO_TWatch_Library/master/examples/Shield/S7xG_LoRa/S7xG_LoRa.ino

incyi, 2020-05-08:
Are you standing outside ?

lewisxhe, 2020-05-09:
Need to be located outdoors, it is best to take the GPS antenna outside the watch, if there is a larger antenna, you can try an external antenna

jsonpoindexter, 2020-05-11:
Thank you for the responses. I am able to get a GPS position after opening the watch and moving the antenna outside the case, as well as leaving the watch outside. Unfortunately it still takes a long time to get a lock. Are there ways to improve this in code? Possibly using hot start (`gpsSetStart(true)`) or not resetting (`gpsReset()`) on initialization? 

Alos, I cannot find documentation for `gps resest`. I am using https://edit.wpgdadawant.com/uploads/news_file/program/2020/35687/tech_files/S7678G_Commands_Set_Reference_1.6.6_G11.pdf

lewisxhe, 2020-05-14:
You can refer to the data sheet to optimize it, if you want to hot start, then the LDO3 power supply can not be turned off, otherwise it will be invalid! You can view the manual here
https://github.com/lewisxhe/Acsip-S7xG-Library/tree/master/docs

jsonpoindexter, 2020-05-15:
@lewisxhe  when does `LDO3` get turned off? Is that every time the watch gets turned off?

lewisxhe, 2020-06-28:
It is connected to the power of S76G. If you don’t want to use S76G, please turn off LDO3 and LDO4. If the problem is solved, please close the issue @jsonpoindexter

closed 2020-06-30 by lewisxhe