Xinyuan-LilyGO/TTGO_TWatch_Library #9: No vibration motor control and backlight flicker

billyischilly, 2020-06-26:
Hi there! I've been trying to get the vibration motor to vary its intensity or frequency for more advanced haptic feedback but it doesn't seem as though calling the motor adjust function or altering the frequency in the bl.h file has any effect. I've also noticed that at certain settings turning the motor on will cause the backlight to flicker or turn off completely. Not sure if this is something that can be fixed or whether it is just a limitation of the device, please let me know! Its a shame as the motor seems relatively powerful and low latency!!

lewisxhe, 2020-06-26:
Unfortunately, the effect of adjusting the intensity of the vibration motor is not ideal, because it does not use a dedicated drive circuit.  The backlight is dark because the vibration consumes too much current.

billyischilly, 2020-06-26:
Ah, I thought it might be something like that! Oh well! Thanks for the quick reply! :)

closed 2020-06-28 by lewisxhe