Xinyuan-LilyGO/TTGO_TWatch_Library #10: MenuBar::lv_menu_config_t register periodic task

miathedev, 2020-06-27:
Hi community,

is there any way to register a periodic task while a "face" is active?
(Which will be closed on exit)

For example i would like to light up the internal LED every second when the following "face" is active:
wifi_event_cb

{.name = "WiFi",  .img = (void *) &wifi, .event_cb = wifi_event_cb}

I need this because I would like to create a simple "compass" app for the SimpleWatch example. So i need to poll the AxisSensor every second.

lewisxhe, 2020-06-28:
You can call
```
lv_task_create
```
,Please refer to [lvgl docs](https://docs.lvgl.io/v6/en/html/overview/task.html)

miathedev, 2020-06-29:
Thank you for your reply.

I already know lv_task_create.
I took a look at the docs anyways and found out i have to refactor the complete code (what i expected...).

The prolem is that the lv_task has to be destroyed after exit (my question was not really clear).

I will create a abstract "face" class that has sth like a destruct and create method.

Faces then will inherit from it.
when im Done i will do a push request.

closed 2020-06-29 by miathedev

lewisxhe, 2020-06-29:
I have upgraded the lvgl in the library to 7.0.0, it is not compatible with the existing version, so please do not submit a request