Xinyuan-LilyGO/TTGO_TWatch_Library #11: Backlight Pin?

Mikumiku747, 2020-06-27:
Hello, 

I'm trying to do control of the backlight on the T-Watch 2020, but the schematic doesn't say which pin controls the backlight, and I can't find it in the backlight code in `src/drive/tft/bl.h`. Is it Pin 12? I also see there is something about the AXP202 LDO2 in `src/TTGO.h`, do I also need to switch that on to turn the backlight on?

Thanks.

lewisxhe, 2020-06-28:
Yes, it is connected to 12, LDO2 it is not only connected to the backlight, but also connected to the LCD driver, if you turn it off, you must re-initialize the driver again when you use it again!

Mikumiku747, 2020-06-28:
Thank you for responding very quickly. I've been enjoying rpogramming with the W-Watch 2020.

closed 2020-06-28 by Mikumiku747