Xinyuan-LilyGO/TTGO_TWatch_Library #17: Playing Audio on the LILYGO_WATCH_2020_V1 breaks the display

jwktje, 2020-07-03:
I saw that the example using earlephilhower/ESP8266Audio wasn't supported on the LILYGO_WATCH_2020_V1.

So I tried using[ this library](https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/src/Audio.h) to play an MP3 from SPIFFS with the following sketch; 

```
#define LILYGO_WATCH_2020_V1
#define LILYGO_WATCH_LVGL 
#include <LilyGoWatch.h>
TTGOClass *ttgo;

#include "Audio.h" //https://github.com/schreibfaul1/ESP32-audioI2S
#include "SPI.h"
#include "SPIFFS.h"
#include "FS.h"

// Digital I/O used
#define I2S_DOUT      33 // TTGO correct
#define I2S_BCLK      26 // TTGO correct
#define I2S_LRC       25 // TTGO correct

Audio audio;

void setup() {
    Serial.begin(115200);
    Serial.println("Hello");
    
    //Init watch
    ttgo = TTGOClass::getWatch();
    ttgo->begin();
    ttgo->openBL(); //Backlight

    //Test for LVGL
    ttgo->lvgl_begin();
    lv_obj_t *text = lv_label_create(lv_scr_act(), NULL);
    lv_label_set_text(text, "T-Watch");
    lv_obj_align(text, NULL, LV_ALIGN_CENTER, 0, 0);

    //!Turn on the audio power
    ttgo->enableLDO3();

    SPIFFS.begin();
    
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(21); // 0...21
  
    audio.connecttoFS(SPIFFS, "/pno-cs.mp3");
}

void loop()
{
    audio.loop();
    lv_task_handler();
    delay(5);
}
```

This works but it makes the LVGL fail. I can see the backlight turning on but nothing is displayed. 
If I remove the `#include "Audio.h"` it does show the text on the screen.

Is there a way to make the LILYGO_WATCH_2020_V1 play Audio from SPIFFS without breaking LVGL?

lewisxhe, 2020-07-04:
I haven’t used the library you said. ESP8266Audio is supported in 2020. You should be blocked, but ESP8266Audio is not blocked, but if you call the GUI, the sound may not be so smooth. The sample program path is ` examples/BasicUnit/Audio/ExternalDAC/external_dac_play/external_dac_play.ino`

jwktje, 2020-07-04:
I didn’t realize that the MAX DAC would qualify as external but that actually makes sense. Does it support using SPIFFS? I use the espressif tool to upload mp3 files. 

https://github.com/me-no-dev/arduino-esp32fs-plugin

lewisxhe, 2020-07-04:
support ..

jwktje, 2020-07-04:
Do you know of any example sketch that shows loading a file from SPIFFS?
Would this work on the LILYGO_WATCH_2020_V1?
 https://github.com/earlephilhower/ESP8266Audio/blob/master/examples/PlayMP3FromSPIFFS/PlayMP3FromSPIFFS.ino

lewisxhe, 2020-07-04:
Please refer to the example, from ram to SPIFFS

closed 2020-07-07 by lewisxhe

jwktje, 2020-07-08:
Sadly this doesn't work. I get these errors: 
```
In file included from /Users/jwktje/Documents/Arduino/Watch_test_v1_mqtt/audio.ino:4:0:
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:32:26: error: expected ')' before '&' token
     AudioFileSourceFS(FS &fs) { filesystem = &fs; }
                          ^
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:33:26: error: expected ')' before '&' token
     AudioFileSourceFS(FS &fs, const char *filename);
                          ^
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:45:5: error: 'FS' does not name a type
     FS *filesystem;
     ^
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:46:5: error: 'File' does not name a type
     File f;
     ^
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h: In member function 'virtual uint32_t AudioFileSourceFS::getPos()':
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:42:47: error: 'f' was not declared in this scope
     virtual uint32_t getPos() override { if (!f) return 0; else return f.position(); };
                                               ^
In file included from /Users/jwktje/Documents/Arduino/Watch_test_v1_mqtt/audio.ino:5:0:
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceSPIFFS.h: In constructor 'AudioFileSourceSPIFFS::AudioFileSourceSPIFFS()':
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceSPIFFS.h:33:55: error: no matching function for call to 'AudioFileSourceFS::AudioFileSourceFS(fs::SPIFFSFS&)'
     AudioFileSourceSPIFFS() : AudioFileSourceFS(SPIFFS) { };
                                                       ^
In file included from /Users/jwktje/Documents/Arduino/Watch_test_v1_mqtt/audio.ino:4:0:
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:29:7: note: candidate: AudioFileSourceFS::AudioFileSourceFS()
 class AudioFileSourceFS : public AudioFileSource
       ^
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:29:7: note:   candidate expects 0 arguments, 1 provided
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:29:7: note: candidate: constexpr AudioFileSourceFS::AudioFileSourceFS(const AudioFileSourceFS&)
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:29:7: note:   no known conversion for argument 1 from 'fs::SPIFFSFS' to 'const AudioFileSourceFS&'
In file included from /Users/jwktje/Documents/Arduino/Watch_test_v1_mqtt/audio.ino:5:0:
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceSPIFFS.h: In constructor 'AudioFileSourceSPIFFS::AudioFileSourceSPIFFS(const char*)':
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceSPIFFS.h:34:85: error: no matching function for call to 'AudioFileSourceFS::AudioFileSourceFS(fs::SPIFFSFS&, const char*&)'
     AudioFileSourceSPIFFS(const char *filename) : AudioFileSourceFS(SPIFFS, filename) {};
                                                                                     ^
In file included from /Users/jwktje/Documents/Arduino/Watch_test_v1_mqtt/audio.ino:4:0:
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:29:7: note: candidate: AudioFileSourceFS::AudioFileSourceFS()
 class AudioFileSourceFS : public AudioFileSource
       ^
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:29:7: note:   candidate expects 0 arguments, 2 provided
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:29:7: note: candidate: constexpr AudioFileSourceFS::AudioFileSourceFS(const AudioFileSourceFS&)
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:29:7: note:   candidate expects 1 argument, 2 provided
Multiple libraries were found for "WiFi.h"
 Used: /Users/jwktje/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
Multiple libraries were found for "SD.h"
 Used: /Users/jwktje/Documents/Arduino/hardware/espressif/esp32/libraries/SD
 Not used: /Applications/Arduino.app/Contents/Java/libraries/SD
exit status 1
Error compiling for board TTGO T-Watch.
```

jwktje, 2020-07-08:
As soon as I comment out the following line it does compile
`#include "AudioFileSourceSPIFFS.h"`

lewisxhe, 2020-07-08:
Ok . I will update the playback example of loading audio files using SPIFFS tomorrow

jwktje, 2020-07-08:
Wow thanks for the super fast reply! It would be awesome if you can help me and make it work!

lewisxhe, 2020-07-09:
Has been added @jwktje

jwktje, 2020-07-09:
Thanks! That example looks very similar to what I was attempting on my machine. Sadly trying to compile the new example I still get errors. Can you compile it successfully? 

```
In file included from /Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceSPIFFS.h:28:0,
                 from /var/folders/2v/6ryy6zr91msf1kwgs5_mwdy80000gn/T/arduino_modified_sketch_14777/PlayMP3FromSPIFFS.ino:12:
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:32:26: error: expected ')' before '&' token
     AudioFileSourceFS(FS &fs) { filesystem = &fs; }
                          ^
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:33:26: error: expected ')' before '&' token
     AudioFileSourceFS(FS &fs, const char *filename);
                          ^
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:45:5: error: 'FS' does not name a type
     FS *filesystem;
     ^
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:46:5: error: 'File' does not name a type
     File f;
     ^
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h: In member function 'virtual uint32_t AudioFileSourceFS::getPos()':
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:42:47: error: 'f' was not declared in this scope
     virtual uint32_t getPos() override { if (!f) return 0; else return f.position(); };
                                               ^
In file included from /var/folders/2v/6ryy6zr91msf1kwgs5_mwdy80000gn/T/arduino_modified_sketch_14777/PlayMP3FromSPIFFS.ino:12:0:
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceSPIFFS.h: In constructor 'AudioFileSourceSPIFFS::AudioFileSourceSPIFFS()':
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceSPIFFS.h:33:55: error: no matching function for call to 'AudioFileSourceFS::AudioFileSourceFS(fs::SPIFFSFS&)'
     AudioFileSourceSPIFFS() : AudioFileSourceFS(SPIFFS) { };
                                                       ^
In file included from /Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceSPIFFS.h:28:0,
                 from /var/folders/2v/6ryy6zr91msf1kwgs5_mwdy80000gn/T/arduino_modified_sketch_14777/PlayMP3FromSPIFFS.ino:12:
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:29:7: note: candidate: AudioFileSourceFS::AudioFileSourceFS()
 class AudioFileSourceFS : public AudioFileSource
       ^
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:29:7: note:   candidate expects 0 arguments, 1 provided
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:29:7: note: candidate: constexpr AudioFileSourceFS::AudioFileSourceFS(const AudioFileSourceFS&)
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:29:7: note:   no known conversion for argument 1 from 'fs::SPIFFSFS' to 'const AudioFileSourceFS&'
In file included from /var/folders/2v/6ryy6zr91msf1kwgs5_mwdy80000gn/T/arduino_modified_sketch_14777/PlayMP3FromSPIFFS.ino:12:0:
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceSPIFFS.h: In constructor 'AudioFileSourceSPIFFS::AudioFileSourceSPIFFS(const char*)':
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceSPIFFS.h:34:85: error: no matching function for call to 'AudioFileSourceFS::AudioFileSourceFS(fs::SPIFFSFS&, const char*&)'
     AudioFileSourceSPIFFS(const char *filename) : AudioFileSourceFS(SPIFFS, filename) {};
                                                                                     ^
In file included from /Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceSPIFFS.h:28:0,
                 from /var/folders/2v/6ryy6zr91msf1kwgs5_mwdy80000gn/T/arduino_modified_sketch_14777/PlayMP3FromSPIFFS.ino:12:
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:29:7: note: candidate: AudioFileSourceFS::AudioFileSourceFS()
 class AudioFileSourceFS : public AudioFileSource
       ^
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:29:7: note:   candidate expects 0 arguments, 2 provided
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:29:7: note: candidate: constexpr AudioFileSourceFS::AudioFileSourceFS(const AudioFileSourceFS&)
/Users/jwktje/Documents/Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h:29:7: note:   candidate expects 1 argument, 2 provided
Multiple libraries were found for "SD.h"
 Used: /Users/jwktje/Documents/Arduino/hardware/espressif/esp32/libraries/SD
 Not used: /Applications/Arduino.app/Contents/Java/libraries/SD
Multiple libraries were found for "WiFi.h"
 Used: /Users/jwktje/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
Using library TTGO_TWatch_Library at version 1.2.0 in folder: /Users/jwktje/Documents/Arduino/libraries/TTGO_TWatch_Library 
Using library SPI at version 1.0 in folder: /Users/jwktje/Documents/Arduino/hardware/espressif/esp32/libraries/SPI 
Using library Wire at version 1.0.1 in folder: /Users/jwktje/Documents/Arduino/hardware/espressif/esp32/libraries/Wire 
Using library FS at version 1.0 in folder: /Users/jwktje/Documents/Arduino/hardware/espressif/esp32/libraries/FS 
[... cut ...]
Using library HTTPClient at version 1.2 in folder: /Users/jwktje/Documents/Arduino/hardware/espressif/esp32/libraries/HTTPClient 
Using library WiFiClientSecure at version 1.0 in folder: /Users/jwktje/Documents/Arduino/hardware/espressif/esp32/libraries/WiFiClientSecure 
Using library ESP8266Audio at version 1.4 in folder: /Users/jwktje/Documents/Arduino/libraries/ESP8266Audio 
Using library SD at version 1.0.5 in folder: /Users/jwktje/Documents/Arduino/hardware/espressif/esp32/libraries/SD 
exit status 1
Error compiling for board TTGO T-Watch.
```

lewisxhe, 2020-07-09:
It seems that SPIFFS is missing from the esp32-croe you installed. You can first use the SPIFFS sample that comes with esp32-croe to see if it compiles normally. If you can, please re-clone the Audio library and try again. I have tested this example  , No problem, works well.

jwktje, 2020-07-09:
The ESP SPIFFS_Test example compiles.
I'm using this: 

- Arduino 1.8.13
- espressif/arduino-esp32 1.0.4

The errors are explicitly mentioning `Arduino/libraries/ESP8266Audio/src/AudioFileSourceFS.h`

Can you post which version of this library you are using? Maybe the newer one I have has some bugs

lewisxhe, 2020-07-10:
You try this, I use this

[ESP8266Audio.zip](https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/files/4900148/ESP8266Audio.zip)

jwktje, 2020-07-10:
Thanks for that! This gave me the info to figure it out. You have version `1.1.3` locally and the repo that I cloned is on `1.4.`. They don't use tags for releases so in order to fix it i had to checkout the [specific commit](https://github.com/earlephilhower/ESP8266Audio/commit/71ec0d590ba94a09afa7df88dca676b6f6470f6a?diff=unified).

I did this like so; `git checkout 71ec0d5`

Now all problems are gone and the examples compile! To verify this on your end you could first; 

`git checkout origin/master` 

then try to compile and see that it (probably) throws errors like I had. And then roll back with; 

`git checkout 71ec0d5`

To avoid confusion in the future maybe you could [use git submodules to add the ESP8266Audio dependency and pin it to this commit](https://stackoverflow.com/questions/10914022/how-do-i-check-out-a-specific-version-of-a-submodule-using-git-submodule) until the library is fixed?

Otherwise if somebody uses the clone commands in the examples like me they will still get these errors. Or alternatively you could mention this issue in the example sketch so people can find the solution

lewisxhe, 2020-07-10:
Okay, I will test the latest version on Monday, and I will fix it by then.@jwktje

jwktje, 2020-07-10:
Thanks! Through testing I found out that the[ first commit ](https://github.com/earlephilhower/ESP8266Audio/commit/65220b8e31e08ec3d4460845e29e753b05b4007b)that introduces this error is `65220b8`

bwagstaff, 2020-09-12:
I found fully scoping them as fs::FS, fs::File, and similar resolved the issues for me.

alastaira, 2023-09-03:
The ESP8266Audio library works fine for me on a v1. However, it took much head-scratching to figure out that was true, because every time I ran HardwareTest.ino it remained silent. I tried all sorts of messing about with SPIFFS and PROGMEM, only to eventually realise that the audio circuit is never actually initialised.... you need to add the following into HardwareTest.ino:

```
//!Turn on the audio power
ttgo->enableLDO3();
```