Xinyuan-LilyGO/TTGO_TWatch_Library #19: Really Simple Instructions? (A call for help)

sparquay, 2020-07-04:
Hello, I'm not really a programmer, but I'd like to be.

I'm really new to all this, and have been trying to learn Python. I thought getting this watch would be a fun project, but it appears I underestimated the difficulty. I've tried looking for some instructional help for a beginner like myself, but as yet haven't found much. I really don't want this to become something I abandon out of frustration either.
I would like some help understanding the very basics. I've got the library installed in Arduino IDE, but I haven't been able to see the watch files on my desktop yet, or even how to push updated software to the watch.
I think the watch came pre-loaded with SimpleWatch... and I've tried to tweak things, but so far, I haven't even figured out how to set the time! 😅
Any help from someone willing to take the little steps with me would be really great. This looks like awesome hardware, and I'd really like to be able to figure out how to use it.

sparquay, 2020-07-04:
[image]
This is as far as I've gotten... I figured out the watch showed up on Port 5 for me... but that's as far as I've gotten. 😅

billyischilly, 2020-07-04:
Happy to help if I can! I wouldn't say I'm a massively experienced coder but I have been messing around with Arduino for about 3-4 years. Do you have a particular thing that you would like the watch to do that you are working towards or just interested in using it as a platform to learn? I would maybe recommend in the latter case starting with a simpler example program and working from there. I think the BMA423_Accel example is quite a good one. It shows you how to take some sensor data from the accelerometer and print some stuff to the screen and not much else - which is good!!! The Simple Watch example actually has quite a lot going on!

sparquay, 2020-07-04:
Yeah, originally, I had grand ideas of different ways to display time (similar to [ToykoFlash](https://tokyoflash.com/) watches), or maybe making a PipBoy like interface (from the FallOut Series). But I should think more simple, as you said. Thanks very much for giving me a starting point.

billyischilly, 2020-07-04:
That all sounds totally doable but yeah, just work your way up to it. When I was first working things out I used the Arduino reference a lot (https://www.arduino.cc/reference/en/) and found the map function very useful for a lot of tasks so maybe look at those. Just work your way through the library examples, trying to understand little bits of code and how you can adapt them to do different things - with the BMA423 example, maybe work out how to draw some shapes to the screen and have their positions/sizes/colours change depending on accelerometer data. The TFT_eSPI->TFT_Ellipse example would be a good one to look at too. That kind of stuff is pretty straightforward but fun and gets you learning and problem solving and will have a lot of relevance when you come to do your Tokyo Flash watch displays and stuff! Hope that's helpful!

sparquay, 2020-07-04:
[image]
More tiny bits of progress... Got something else on the watch. I did get quite a laugh by this display, even though I don't know what the Chinese says... @lewisxhe ?

lewisxhe, 2020-07-04:
@sparquay  I’m glad there are more people involved. I’m working hard to maintain this library so that it doesn’t look so bad, but LilyGo is constantly making new boards and I have to allocate time to other boards. I try to keep simple sample programs to make it easier for novices. For 2020, I will take the time to introduce which examples can be run on which boards, without letting novices make unnecessary guesses. Thanks again to @billyischilly for your answer, and your first picture shows that you are using the previous version of the library. I have rearranged this library in the past few days and upgraded lvgl to 7.0. Your attempts and feedback are welcome. .

Just to say that, the meaning in the picture is to say "speak some time",

sparquay, 2020-07-04:
Apologies for making it untidy, but it was nice to have a bit of community feedback. Is there another place these sort of discussions would be better? I understand you're a busy guy @lewisxhe , but somehow, I'm becoming a fan as well! 👍

lewisxhe, 2020-07-06:
@sparquay There is a group of lilygo on facebook, you can look there, software questions are best asked on github.

closed 2020-07-07 by lewisxhe