Xinyuan-LilyGO/TTGO_TWatch_Library #20: Writing to the display from multiple FreeRTOS tasks causes watchdog failure

red-fox-star, 2020-07-04:
Thanks for this watch. I’ve been using it as a way to learn FreeRTOS and enjoying it. I recently broke my code up into multiple tasks and the watch becomes unstable, crashing with watchdog errors. 

I believe the reason is because the display code is getting preempted between the [begin transmisson](https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/blob/master/src/drive/fx50xx/FT5206.h#L114) and end transmission lines, causing a thread to never get its response. 

Does it make sense to add a mutex around those lines? I’d be happy to send a pull request if so.

lewisxhe, 2020-07-04:
Among multiple tasks, I suggest that you place the display task in the main task, rather than in multiple tasks, and run tasks that are not related to display in other tasks.

red-fox-star, 2020-07-04:
@lewisxhe I considered that, but that would complicate the structure of the code considerably. Defending the port in the driver would allow whatever code architecture someone wants to use to work without issue rather than requiring a single-thread display pattern. 

If it’s not welcome to add a mutex to the display code here, I’ll implement it as a global or possibly just run my own fork of the library.

closed 2020-07-04 by red-fox-star