Xinyuan-LilyGO/TTGO_TWatch_Library #24: How do you get the T-Watch 2020 to go into a mode in which it can be programmed

G4DCP, 2020-07-07:
All attempts at loading a sketch fail timeout.
What do you do to put in boot mode?

G4DCP
Peter D Hull

KennyM64, 2020-07-08:
1. I had a couple fail on me. I made sure that the watch was on (not sleep mode). Push the button for 2 seconds. Then whenever I upload it works. I presume that was the problem for me.

2. Are you commenting out this line in the config.h file?
 #define LILYGO_WATCH_2020_V1             //To use T-Watch2020, please uncomment this line

3. You've downloaded the latest board information and select the "TTGO T-watch" as your board in the arduino program.

4. You've selected the correct port.

Hope this helps :-)

lewisxhe, 2020-07-08:
@G4DCP As @KennyM64 said, the first point is necessary, you cannot download it without booting. The second and third point, if the wrong choice is ultimately an exception of the program, it will not say that the download failed

G4DCP, 2020-07-08:
I have now got an exmple to load.
I had not held the button for 2 seconds,.

Thanks Kenny and Lewis. I was aware of the Define in config.h and I had the correct port but I missed hold the button for 2 seconds at the right time!

Peter

closed 2020-07-08 by G4DCP

funnypolynomial, 2020-07-24:
I can't get anything to load. Compiles fine (ignoring warnings) but loading fails:
```
esptool.py v2.6
Serial port COM4
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
```
Any ideas?  If I turn the thing on just before (2second press on crown) the display turns off when Connecting... starts. Which makes me think maybe it's **trying** to re-program.

KennyM64, 2020-07-25:
Are you sure you've got the right port & board selected?

KennyM64, 2020-07-25:
It should go dark when uploading so it does look like it's trying to load. 

Have you uncommented the line "#define LILYGO_WATCH_2020_V1 " in the config.h file?
Try a different cable or test that the cable works uploading to a different board.

The message that you should get on a successful upload is this.

Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: f0:08:d1:c9:0b:40
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 2000000
Changed.
Configuring flash size...
Auto-detected Flash size: 16MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 16384.2 kbit/s)...
Hash of data verified.
Flash params set to 0x024f
Compressed 15856 bytes to 10276...

Writing at 0x00001000... (100 %)
Wrote 15856 bytes (10276 compressed) at 0x00001000 in 0.1 seconds (effective 1566.0 kbit/s)...
Hash of data verified.
Compressed 326144 bytes to 154267...

Writing at 0x00010000... (10 %)
Writing at 0x00014000... (20 %)
Writing at 0x00018000... (30 %)
Writing at 0x0001c000... (40 %)
Writing at 0x00020000... (50 %)
Writing at 0x00024000... (60 %)
Writing at 0x00028000... (70 %)
Writing at 0x0002c000... (80 %)
Writing at 0x00030000... (90 %)
Writing at 0x00034000... (100 %)
Wrote 326144 bytes (154267 compressed) at 0x00010000 in 2.7 seconds (effective 961.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.0 seconds (effective 4915.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

funnypolynomial, 2020-07-25:
I have #define LILYGO_WATCH_2020_V1
Board is "TTGO T-Watch"
On COM4 ("Silicon Labs CP210x USB to UART Bridge")
Everything else is at default (Upload speed "2000000" etc)
I've tried other cables. Other speeds etc. Running out of things to try :-(
Note that I get the -DARDUINO_T-Watch compile-time warnings ("whitespace after the macro name")

KennyM64, 2020-07-26:
Hm, I don't get the same warning. 
Is it possible that there is a conflict with a previous version of TWatch? 
Is it possible to try it on a clean computer? Just load Arduino, ESP32 board through board manager and TWatch as per the install instructions here https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library.

Sorry but this is my best guess. If it doesn't work maybe give Lilygo support a shout.

funnypolynomial, 2020-07-26:
Thanks. I'll try on another machine.  Starting to think I have a bad unit.
Re the library install... 
I just do the "Download a zipfile from github...Add .ZIP Library..." step, right?
The other step, "Clone this git repository..." is an alternative, or something?

KennyM64, 2020-07-26:
The easiest IMO is to add .zip through Sketch->Include Library>Add .ZIP Library. 

Cloning through GitHub is used to keep up to date with changes and add changes to the Github yourself. It has a lot of features relating to highlighting changes etc. Basically a source controller. When using Git to download and open projects in GitHub desktop it wants to use  Visual Studio Code (well on my system anyway) and it doesn't compile for me. If I use VSC I need to import as an Arduino project so as to set the right paths for libraries etc. 

So that's why I recommend the .ZIP library approach if you're just messing around with examples.

funnypolynomial, 2020-07-27:
**Success!**  I got it to upload using my work desktop machine, pretty much a clean environment, installed 1.8.13 IDE, added .ZIP library, added esp32 board, picked an example (BatmanDial), edited config, hit Upload.  All good!
Very different from my home laptop environment which has had a series of IDE versions installed over the years, plus libraries etc.  Apparently it's become a bit of a mess behind the scenes.  But I can work backwards from this, maybe remove/uninstall then start from scratch (shudder).

Thanks for your help!

KennyM64, 2020-07-27:
No problem, glad I could help. Version incompatibilities are a minefield and are really difficult to sort out. That's where a clean machine with a fresh install can help in the diagnosis of the issue.

Have fun with the watch. :-)