Xinyuan-LilyGO/TTGO_TWatch_Library #26: T-Watch 2020 wont charge batttery

bloutix, 2020-07-08:
Hi,

after many test mm ttwatch 2020 battery won't charge?

How to fix that?

thanks

lewisxhe, 2020-07-08:
Do you get charging current?  How do you judge it cannot be charged?

bloutix, 2020-07-08:
I have tried AXP20x samples and battery voltage is around 2.9V and vbus is around 120mA.
I have tried to set charge current between 300 and 1000mA with no success.
I have tried on usb port, on usb wall charger, i replaced the battery. All without succes.
The battery percentage stay at 0% .

lewisxhe, 2020-07-09:
@bloutix Please run the AXP202 monitoring example, you can see whether it is normal, if not, please contact customer service

https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/blob/master/examples/BasicUnit/AXP20x_ADC/AXP20x_ADC.ino

bloutix, 2020-07-10:
The monitoring example display this values:
Vbus: 5057.50mV
Vbus: stay beetwen 75mA and 131.62mA
BATT: 2960.10mV
Per: 0%

lewisxhe, 2020-07-10:
Can you tell me the complete display? Or give me a photo

bloutix, 2020-07-10:
[cid:12150f3d-acf1-47f4-8060-bb5933b360b4]

lewisxhe, 2020-07-10:
Please update the warehouse to the latest, it contains more detailed information

bloutix, 2020-07-10:
[image]

lewisxhe, 2020-07-10:
It looks like the battery is connected. You can unplug the USB, leave the battery in a discharged state, then wait for tens of minutes to let the power drain, and then plug in the USB again to check its charging current

bloutix, 2020-07-10:
With the original battery charge is at 0.00mA and the bat voltage decrease slowly (around 2800mV),
with another battery charge is 0mA or sometime it's at 2mA with a bat volatge around 3270mV

lewisxhe, 2020-07-10:
It seems that the charging part may be damaged, please contact LilyGO sales, this can not be repaired by the software.

closed 2020-07-10 by lewisxhe