Xinyuan-LilyGO/TTGO_TWatch_Library #27: Pinout details

sdrabb, 2020-07-08:
Please provide us esp32 pinout connection details.

lewisxhe, 2020-07-09:
LilyGo does not provide the schematic diagram of TWatch, all IO Pins are defined in the file

https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/blob/master/src/board/twatch2020_v1.h

closed 2020-07-09 by lewisxhe

NilsBe, 2020-07-10:
I understand that schematics are not published.
Would it be possible though to give some details about the connected pins of the AXP202?

I would love to understand this a little better:

    // Turn off unused power
    ttgo->power->setPowerOutPut(AXP202_EXTEN, AXP202_OFF);
    ttgo->power->setPowerOutPut(AXP202_DCDC2, AXP202_OFF);
    ttgo->power->setPowerOutPut(AXP202_LDO3, AXP202_OFF);
    ttgo->power->setPowerOutPut(AXP202_LDO4, AXP202_OFF);

And what about: 

AXP202_EXTEN
AXP202_DCDC3
AXP202_LDO2
AXP202_LDO4
AXP202_DCDC2
AXP202_LDO3

stackshadow, 2020-07-16:
Found it in the docs. See here:
https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/blob/master/docs/pinmap.md#3-axp202-power-management-channel

tuupola, 2020-10-07:
@stackshadow There are some gotchas such as just enabling `LDO2` is not enough. You also need to enable PWM on `GPIO12`. I do not really understand why. Still work in progress but I am documenting my findings at:

https://github.com/tuupola/esp_twatch2020

drunkscientist, 2021-02-28:
> LilyGo does not provide the schematic diagram of TWatch, all IO Pins are defined in the file
> 

pin numbers are NOT a substitute for a schematic or wiring diagram, it would be helpful if they were released
@tuupola do you happen to know if the unused LDO4/LDO5 are enabled in software? like if i take some tiny wire and tiny soldering iron with a microscope, will a program calling them work or throw an error?

lewisxhe, 2021-03-01:
LDO4,5 are not used in V1.

destroyedlolo, 2021-03-01:
Hello,

TTGO kindly provided the schema and I published it on my website with their agreement :
http://destroyedlolo.info/ESP/Tech%20TWatch/

I got the same for the LilyPI but didn't plushed it.

Bye

Laurent

drunkscientist, 2021-03-02:
@destroyedlolo this is AWESOME THANK YOU

X-Ryl669, 2021-11-17:
@destroyedlolo Can you also publish the LilyPI schematic too ?