Xinyuan-LilyGO/TTGO_TWatch_Library #30: is it possible to wake up the TWatch 2020 from deep sleep by pressing the button.

NilsBe, 2020-07-10:
Everything already in the title. 

I checked and uploaded the deep sleep sketch. 
It's great and it might be necessary to use the deep sleep mode for a reasonable battery run time. 

But IMHO the deep sleep makes only sense if one can wake it up by pressing the button.

As schematics are not available I would be really greateful an example sketch for such a solution.

Thanks a lot.
Nils

xaksh, 2020-07-10:
Following this.. would be really helpful to get insight

sdrabb, 2020-07-10:
Deep sleep cause esp32 reboot, i think is a bit reasonable esp32 light_sleep

lewisxhe, 2020-07-10:
Most of them are early examples, and there is no adaptation for 2020.I will update the deep sleep example on Monday.

closed 2020-07-10 by lewisxhe

NilsBe, 2020-07-10:
Fantastic! Great work Lewis.
Thanks.