Xinyuan-LilyGO/TTGO_TWatch_Library #35: Backlight dims when calling shake()

sweetlilmre, 2020-07-11:
After calling the shake() function the backlight dims.
To reproduce, replace the function setting_event_cb() in gui.cpp in the SimpleWatch example with

```
static void setting_event_cb()
{
  TTGOClass *ttgo = TTGOClass::getWatch();
  ttgo->motor_begin();
  ttgo->shake();
  menuBars.hidden(false);
}
```

lewisxhe, 2020-07-13:
Does the backlight dim after shake? It should recover after shaking.

sweetlilmre, 2020-07-13:
Hi, it dims immediately after shake and does not recover.

lewisxhe, 2020-07-14:
I will check if I have time! I will record this question and the question will be closed!

closed 2020-07-14 by lewisxhe

sweetlilmre, 2020-07-14:
Thank you, I appreciate it. I will also look when I have time

billyischilly, 2021-11-09:
Hi! Was this ever fixed or does anyone know how to manually reset the screen brightness after it dims? I'm still using an older release of the library (1.2) with the V1 watch and can't seem to get full brightness. back except by rebooting.

sweetlilmre, 2021-11-09:
It doesn't look like it and development seems to have stalled here.

The motor seems to be directly controlled via IO (https://github.com/destroyedlolo/TTGO_TWatch_Library/blob/master/docs/watch_2020_v1.md). I'm assuming that there is some kind of direct IO write that is interferring with another peripheral (most likely the AXP202) although there could be something else completely out of band happening. If I ever get the time to look at this again, I'll attempt to find a solution.

sweetlilmre, 2021-11-09:
Excellent! If possible, would you submit a pull request to this repo and mention in a following comment so that others can see this solution?

billyischilly, 2021-11-09:
Sure! Hope I've done this right - only my second ever pull request!

#146

sweetlilmre, 2021-11-11:
Looks good to me. I assume you changed the chanel to 4 so that backlight is 0, buzzer 2 and motor 4, giving a channel separation for each function?

@lewisxhe could you please take a look at this?

billyischilly, 2021-11-11:
Yeah, haven't tested anything with the buzzer yet but realised I might have created more problems on channel 3. All seems to still be working fine so far!