Xinyuan-LilyGO/TTGO_TWatch_Library #38: Documentation for upgrading the tft library

red-fox-star, 2020-07-13:
The TFT_eSPI library bundled within the watch library makes it convenient to use with minimal configuration, but it makes it difficult to know how to use a more recent version. The current library version is 2.2.14 and includes some bug fixes since 2.2.6. I hoped it'd be as simple as specifying 2.2.14 as the TFT_eSPI version, and copying over the [setup header](https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/commits/master/src/TFT_eSPI/User_Setups/Setup45_TTGO_T_Watch.h) file, but that left me without a working display.

Can you share what other steps are taken to make the display work with the library?

lewisxhe, 2020-07-14:
Updating TFT_eSPI is more troublesome, because TFT_eSPI uses absolute path writing, direct replacement is not feasible, there are many steps to change. Can you tell me what has been updated in TFT_eSPI? Is there any obvious improvement to T-Watch?

red-fox-star, 2020-07-14:
Nothing specific in the latest releases. I was discussing some changes to the way the text is written to the screen with the author and wanted to be on the latest version of the library. 

The reason I asked for upgrade documentation instead of an upgraded library is that I don’t think it’s _necessary_ to upgrade.

lewisxhe, 2020-07-15:
As I said, the upgrade requires a very tedious copy and paste, not a document can explain, you can refer to how the source code is modified, for those who are already familiar, use TFT_eSPI + AXP202_Library directly, you can complete, no It is necessary to use this library, this library is only for some novices

closed 2020-07-15 by lewisxhe