Xinyuan-LilyGO/TTGO_TWatch_Library #39: Doesn't work with platformio

ardera, 2020-07-13:
Hello,

none of the LVGL examples work with platformio. Neither when using the officially registered [TTGO TWatch Library](https://platformio.org/lib/show/6875/TTGO%20TWatch%20Library) nor when using the this repo as the library.

My `platformio.ini` currently looks like this:
```ini
[env:ttgo-t-watch]
platform = espressif32
board = ttgo-t-watch
framework = arduino
lib_deps =
	https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library
```

And I am currently using an unmodified version of the `examples/LVGL/SimpleWatch` example. I also tried the `Lvgl_Base` example and that didn't work either.

This is the output of the serial monitor:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=0, scl=1
ENTER IN LIGHT SLEEEP MODE
```

sdrabb, 2020-07-13:
@ardera try updating library and  use Arduino env.

Let me know if problem ramains

ardera, 2020-07-13:
> try updating library and use Arduino env.

My library is on latest master. What do you mean by Arduino env? The arduino code editor is not that nice to work with, I'm happy I was able to uninstall it.

sdrabb, 2020-07-13:
i've just tried with Arduino ide and it works, try with it.

sweetlilmre, 2020-07-14:
Work with Platformio for me. What I did was:
1. VS Code install
2. Install platform IO
3. Import Arduino project: SimpleWatch (check the Use Arduino Libraries)
4. Build
5 Upload

Sample project was copied out of the examples folder into c:\dev\arduino\SimpleWatch first.
and config.h edited appropriately

lewisxhe, 2020-07-14:
Platformio works without any problems, I have been working in platformio. From the log you provided, this should be the i2c bus is occupied. What version of Twatch do you use?

ardera, 2020-07-14:
Okay, seems like this was actually my fault. I can't reliably reproduce the issue anymore. Maybe it was due to some global libraries, I'm not sure. Anyway, it works fine now when I use this repo as a `platformio.ini` dependency.

If I hit the issue again, and find that it's caused by the twatch library, I'll open a new issue

closed 2020-07-14 by ardera