Xinyuan-LilyGO/TTGO_TWatch_Library #41: Powering Modules and Proofreading simple App

BioBox, 2020-07-15:
Out of all the examples given, `SimpleWatch` is the most complex. One segment of code that really caught my attention is the following:
```
// Turn off unused power
    ttgo->power->setPowerOutPut(AXP202_EXTEN, AXP202_OFF);
    ttgo->power->setPowerOutPut(AXP202_DCDC2, AXP202_OFF);
    ttgo->power->setPowerOutPut(AXP202_LDO3, AXP202_OFF);
    ttgo->power->setPowerOutPut(AXP202_LDO4, AXP202_OFF);
```
What exactly do each of these AXP registers power? If I'm not going to use a certain module from the watch, is there a way of finding out which AXP register to disable?

Secondly, I want to write a sketch so that the watch will vibrate every 2, 3, or 5 minutes that is cycled everytime you push the button.
```
void setup() {
	Serial.begin(115200);

	ttgo = TTGoClass::getWatch();
	ttgo->begin();
	ttgo->openBL();

	pinMode(RTC_INT, INPUT_PULLUP);
    attachInterrupt(RTC_INT, [] {
        rtcIrq = 1;
    }, FALLING);

    ttgo->rtc->disableAlarm();
    ttimes_len = sizeof(ttimes)/sizeof(ttimes[0]);

    ttgo->tft->fillScreen(TFT_BLACK);
    ttgo->tft->setTextColor(TFT_WHITE, TFT_BLACK);
    ttgo->tft->setTextFont(4);
    ttgo->tft->drawString("Timer is OFF", MESSAGE_X, MESSAGE_Y);

    ttgo->button->setClickedHandler(clicked);
}

void loop() {
	if (rtcIrq) {
		vibrate();
	}
	ttgo->button->loop();
	if (millis() - millis_click > DISPLAY_CD) {
		ttgo->displaySleep();
	}
	delay(50);
}
```
The clicked handler wakes and updates the display, clears and resets the timer, and removes the interrupt request. I'd like to know that this code is correct and stable, because I've never done embedded systems before and I don't want to break the thing using bad software. There's also very little documentation so I had to look through a lot of the drivers and datasheets of the chips. I just want to make sure I know what I'm doing. Thanks.

lewisxhe, 2020-07-16:
SimpleWatch was written when T-Watch was first established. At that time, I was also a newbie in lvgl, so I wrote it more redundantly.
Regarding the AXP power supply channel, except that LDO3 is used on a common backplane, the remaining channels are not used. If you are using a standard backplane, then LDO3 will control the vibration motor power supply, and the rest can be set to off. I will add a description of the power channel in the README.

In the code you submitted, it seems that you are going to use RTC_IRQ for timed vibration, but you have not set an alarm. I think you should refer to the sample code below.

lewisxhe, 2020-07-16:
https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/blob/master/examples/Shield/AlarmClock/AlarmClock.ino

It is a simple demonstration of how to use the alarm clock function.

BioBox, 2020-07-16:
I don't want it to vibrate at a specific time, but at specific time **intervals**. The clicked handler uses the timer function `ttgo->rtc->setTimer(ttimes[ttimes_index], 0x00, false);` not the alarm. 

Judging from the datasheet of the chip, you can also use the timer for RTC_INT.
[image]

From section 8.8 it says:
> At the end of every countdown, the timer sets the timer flag TF. [...]  The asserted TF can be used to generate an interrupt on pin INT.

So `setTimer` should cause an RTC_IRQ too right?

lewisxhe, 2020-07-16:
Yes, it will also trigger RTC IRQ

BioBox, 2020-07-16:
Does the 2020 model use the standard backplane?
It looks like the only ones that I need are LDO 2 & 3, so I can turn the rest of them off.

Thank you very much I'll let you know if it works.

lewisxhe, 2020-07-16:
In T-Watch 2020, LDO3 is the audio power domain, and the remaining power channels are not applicable

closed 2020-07-16 by BioBox