Xinyuan-LilyGO/TTGO_TWatch_Library #42: Schematics

Testato, 2020-07-17:
This is a developer watch.
The best thing would be that the project was open source, both hw and sw.
We can understand that maybe you don't want to provide the pcb project files, but at least the wiring diagram is absolutely necessary.
It is not possible to develop sw for an hw object safely without having at least the schematics

lewisxhe, 2020-07-18:
LilyGo does not provide schematic diagrams. If you really need it, you can try to contact LilyGo by email

closed 2020-07-18 by lewisxhe

Testato, 2020-07-22:
No, this is the answer to my email:
"Sorry, our schematics are not open to the public."

So this watch is not for developers?  We believe it is a serious commercial mistake.