Xinyuan-LilyGO/TTGO_TWatch_Library #50: Instructions to use Arduino IDE on Mac OS X

fcohen, 2020-07-22:
Are there instructions for developing on Arduino IDE on Mac OS X (Catalina)? I installed the TTGO_TWatch_Library and see under File -> Examples the TTGO examples are listed as INCOMPATIBLE. Also, I'm not sure what I should be doing to compile for ESP32? -Frank

Altron-2k20, 2020-07-22:
Have you installed the additional board definitions ?
Menu Arduino -> Preferences -> Additional Boards Manager -> Insert the following URL: https://dl.espressif.com/dl/package_esp32_index.json

fcohen, 2020-07-22:
Thank you, I got it to work! :-) Arduino IDE complains of INCOMPATIBLE examples until I 1) Installed the ESP32 dev tools, 2) Installed the ESP32 board definitions, 3) Updated the ESP32 board location in the Arduino IDE Preferences. And now I changed config.h for the Batman example, chose the TTGO T1 board from the Tools -> Board menu in IDE, selected the correct USB port, and clicked Upload. Thanks for your support.

fcohen, 2020-07-22:
Does TTGO do custom product development? I need a watch product built to my designs, including adding GPS to the base design of the TWatch-2020. -Frank

lewisxhe, 2020-07-23:
You need to choose T-Watch instead of T1. Their flash memory capacity is inconsistent. For customization, please contact LilyGo customer service.

closed 2020-07-23 by lewisxhe