Xinyuan-LilyGO/TTGO_TWatch_Library #53: documentation unclear

bferrell, 2020-07-24:
Where does the config.h file go?

lewisxhe, 2020-07-24:
`all examples include a config.h file` Is this sentence unclear?

bferrell, 2020-07-24:
Lewis,  I did not find a config.h in the TTGO watch 2020 examples in the arduino IDE, hence my question

lewisxhe, 2020-07-24:
Check out the examples, each example contains

KennyM64, 2020-07-26:
When you open a TTGO example (.ino) file it should appear in the IDE. There should be another "TAB" that says config.h. Click on this tab and change the setting. Sometimes there are multiple tabs and you have to click the down arrow in the IDE to expose all the tabs in a drop-down list and then select the config.h.
[image]

bferrell, 2020-07-26:
> lilygo <https://user-images.githubusercontent.com/30052801/88471442-11b0ac80-cf3c-11ea-94d9-2d077633e3cb.PNG>

Very helpful!  Thank you!

closed 2020-07-29 by lewisxhe