Xinyuan-LilyGO/TTGO_TWatch_Library #56: Rendering issues when switching image source

wizche, 2020-07-27:
Howdy,

I created a small mario bros animation where the character jumps at regular interval. See the following video:
[image]

As you can see in the video, from time to time, the image gets dirty and white pixels are rendered around the character. 
Any idea why this is happening?
The code for rendering mario can be found here: [https://github.com/wizche/ttgo-smb/blob/75df0a7e8fb3011f1353eb3817e7397935f53be9/src/mario.cpp#L14](https://github.com/wizche/ttgo-smb/blob/75df0a7e8fb3011f1353eb3817e7397935f53be9/src/mario.cpp#L14)

Another question, since I set background transparency on the main screen object ([see here](https://github.com/wizche/ttgo-smb/blob/75df0a7e8fb3011f1353eb3817e7397935f53be9/src/gui.cpp#L8)), why do I need to apply it again to all the container added to the scene? Shouldn't be inherited?

Thanks a lot!

lewisxhe, 2020-07-29:
1. You can go to lvgl[community](https://forum.lvgl.io/) for help for the problem you mentioned
2. Set the main screen to be transparent, and it will not be inherited from other newly created containers. If it is correct, I remember that the copied styles need to be passed when creating a new container to be inherited. Check https://docs.lvgl. io/v7/en/html/overview/style.html

wizche, 2020-07-30:
For reference here the cross-posted issue https://forum.lvgl.io/t/rendering-issues-when-switching-animating-image-source/2866

KennyM64, 2020-07-31:
Have you tried different color formats when converting the image? It's a long shot but the transparency from the png may be going haywire based on the format.

I've downloaded the simulator for LVGL which makes the coding process a lot faster and can highlight hardware problems through screen comparisons from the PC to the watch.

Having said that I too seem to have an issue with dropped pixels. I've put it down to either my code, as this is a fairly new language/library to me or the transparency going bad from the format conversion. If I stumble across anything I'll let you know.

You can see the second hand losing it's tip around 6 and in other places. [https://youtu.be/DZiXT8S8NnQ](url)

wizche, 2020-07-31:
I used true color with alpha, which is AFAIK the only encoding supported with alpha channel integrated right? I guess if the image encoding is the problem then it should happens regularly...

What are u using as simulator? I've setup platform.io in vscode but didnt ported the code there yet. But its a good suggestion, if the problem is not reproducible in the simulator it may be hardware related.

Btw nice job with the Mickey Mouse! Love the watch face :+1: Looking forward to the source release :)

KennyM64, 2020-07-31:
Hi Wizche, like I said it's a long shot but something I was going to rule out myself. It seems to be a transparency issue so trying different formats will rule out a format issue. 

I use Visual Studio 2019 with the LVGL [simulator.](https://github.com/lvgl/lv_sim_visual_studio_sdl). This GIT is designed for 2017 so there was a bit of shoehorning to get it to work. From memory, the wizards took care of most of the translations but there was some issue which I can't recall but I was able to circumvent. If you come across any issues let me know as it might jog my failing memory.

You can see from this [vid](https://www.youtube.com/watch?v=meDTDMXL6VA) that Mickey works well in the simulator as opposed to the watch. I'm also able to through some debugging messages and additional functions to help with debugging. Also the LGVL VS simulator contains examples of all their widgets and a heap of example code. While you have to port the end program to the TTGO environment, most of it is LVGL code, so it's pretty easy. The main code changes are system functions (Date/Time) and event capturing.

Thanks :+1: . I think Mickey looks cool and it's an apple type face so it looks like I have $300 watch that only cost $50. Looking at getting the BLE to work as a Beacon and turn my lights on/off through presence detection on my OpenHab install. I've got it to work with other ESP32 modules so it shouldn't be too difficult. All great fun. :-)

wizche, 2020-08-02:
I dont see any other format supporting alpha channel in the [online converter](https://lvgl.io/tools/imageconverter), what format were u thinking of? I really dont want to write my own image decoder...

I tried the simulator for [vscode/platformio](https://github.com/lvgl/lv_platformio) as well but on MacOS I couldn't get it running (just building). Any chance u can try my sample on ur simulator? I really suspect that it will work as well... 
I guess the easiest fix for me would be to remove the alpha layer and just use as background color the same blue as the background. But thats a pretty sad fix... 

Maybe some maintainers will jump in into the discussion?

KennyM64, 2020-08-03:
OK, my bad. I thought there were more formats with alpha channels, but there aren't.

I had a look at your code and tried to port it to the simulator. Unfortunately, it's too much Arduino entwined code for a quick and dirty check in the simulator. Sorry, I just don't have the time. You're also using animation which I haven't gotten in to yet. My hands transform with the image pivot and angle properties so it's completely different from your setup.

When you say you got the program to build in VS but couldn't get it to run, what did you mean? Have you ported the code and got it to build with no errors? If so send me the project and I'll run it on my VS. I just didn't have time to port your code but if you have it ported and compiling I'll try and run it and send you a vid of the output.

wizche, 2020-08-20:
I created a minimal firmware showing the issue: https://github.com/wizche/flickering-mario
The demo app switches the image each second, we see that at every switch some pixels stay dirty from previous bitmap.,

wizche, 2020-09-06:
Looks the problem is solved using the latest version of the library, maybe b7b062fa103dd9e328b529e18bb787064366adb7 fixed it? 🤔

closed 2020-09-06 by wizche