Xinyuan-LilyGO/TTGO_TWatch_Library #57: Is there an example of over-the-air program loading/booting?

fcohen, 2020-07-30:
I'd like to upload and run an Arduino IDE sketch over the TWatch ESP32 Wifi or BLE connections to my Mac laptop.

KennyM64, 2020-07-31:
All the OTA examples in the Arduino IDE (Web & IP) worked fine for me. A good IP one to start with is File>Examples>Examples for TTGO T-Watch>BasicOTA

Just integrate the example code into the TWatch application that you've written and it's then updatable Over The Air. :-)

fcohen, 2020-07-31:
It works! Thanks.

BTW, on a Mac, the BasicOTA is in 
/Users/frankcohen/Documents/Arduino/hardware/espressif/esp32/libraries/ArduinoOTA/examples/BasicOTA

And instructions I found at
https://lastminuteengineers.com/esp32-ota-updates-arduino-ide/

-Frank

incyi, 2020-08-04:
Great idea ! I’ll see if I can do something with this.

lewisxhe, 2020-08-27:
It looks like it has been resolved, the issue will be closed

closed 2020-08-27 by lewisxhe

YordanYanakiev, 2021-06-13:
I am unable to find reliable example as well.
Using the "updater" example works on ESP32 Devkit 1, and a few other ESP32 boards I have in home, but it does not works with the TTGO T-Watch 2020 v1 I have ( I have 10 pieces currently ).
So - is there some actually tested and working OTA example for autoupdating from a Web server ?
Something like the Arduino's example with **HTTPUpdate.h**
"arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\TTPUpdate\examples\httpUpdate" ?