Xinyuan-LilyGO/TTGO_TWatch_Library #58: SimpleWatch crashes with Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

KennyM64, 2020-07-31:
SimpleWatch has worked previously and I haven't changed any code. When I compile and upload it says upload complete and screen remains dark. Opening the serial monitor reveals this. What's going wrong?

Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0xfefefefe  PS      : 0x00060f30  A0      : 0x800d6e14  A1      : 0x3ffb1e70  
A2      : 0x00000053  A3      : 0x3ffb1edd  A4      : 0x00000001  A5      : 0x3ffb86ec  
A6      : 0x00000000  A7      : 0xffffff01  A8      : 0x80166ef0  A9      : 0x00000053  
A10     : 0x000000fe  A11     : 0x00000053  A12     : 0x3ffb1edd  A13     : 0x00000001  
A14     : 0x00000051  A15     : 0x00000000  SAR     : 0x00000001  EXCCAUSE: 0x00000014  
EXCVADDR: 0xfefefefc  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x7efefefe:0x3ffb1e70 0x400d6e11:0x3ffb1e90 0x400d71e5:0x3ffb1ed0 0x400d6b48:0x3ffb1f00 0x400d1edb:0x3ffb1f30 0x400fa5e3:0x3ffb1fb0 0x4008d3cd:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac

lewisxhe, 2020-07-31:
Please make sure to select the correct model in config.h, and select the correct Board in ArduinoIDE

KennyM64, 2020-07-31:
It is correct as I've uploaded this before. Double checked the config.h #define LILYGO_WATCH_2020_V1. Check. Port is correct as I upload other sketches to this port all the time and directly after the fail.

lewisxhe, 2020-07-31:
Have you updated Lib recently? SimpleWatch I run the example is working fine

KennyM64, 2020-07-31:
Updated 23/7/20. As I said it worked fine before but now doesn't. It may
have something to do with the LVGL version as I did upgrade to 7.4 to try
and get image rotate to work. It should however be backwardly compatible so
I don't think this is the issue. Also, the compile should be taking the 7.0
release of LVGL that is included with your release as I'm opening and
compiling from a separate folder, not installing the libraries to Arduino.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

KennyM64, 2020-07-31:
I've just downloaded the git as a zip and unzipped it to a new folder. Loaded SimpleWatch from the new download and edited the config.h file. Compile and upload. Everything comes back ok and the watch does a hard reset and then nothing. Open the serial monitor and I can see the board looping continuously on the Guru Meditation Error: 

I've also done a hard reset of my computer and it still persists. 

I will try it on another computer when I get a minute.

KennyM64, 2020-07-31:
Sorry, my fault. I just tried it another computer and it works fine. I must have a version issue on my main dev computer. I'll track it down. ;-)

closed 2020-07-31 by KennyM64