Xinyuan-LilyGO/TTGO_TWatch_Library #63: sketch cannot be uploaded

rctron, 2020-08-19:
What could be the problem? The watch new,original firmware working.. 
Arduino 1.8.9. (https://dl.espressif.com/dl/package_esp32_index.json) ESP32 driver ok,TTGO T-Watch ok,sketch compile ok.
Upload - then error :
esptool.py v2.6
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
(on-off button pressing does not help ..)

lewisxhe, 2020-08-20:
Have you turned on the power, programming is not possible when it is turned off, press and hold the Twatch button for two seconds and release it, and then try to upload again

rctron, 2020-08-20:
Thanks for the reply.,it didn't help. I get this after pressing a button 2sec : (terminal screen) : 
[image]
Any other ideas?

lewisxhe, 2020-08-27:
This looks like it is already on, can you still not be able to program?

Please try to lower the baud rate, if it still does not work, please contact LilyGo customer service, it looks like an automatic download failure

closed 2020-08-27 by lewisxhe