Xinyuan-LilyGO/TTGO_TWatch_Library #70: how to save factory firmware for secure ?

picpic020960, 2020-09-04:
Bonjour ,
i want to backup factory firmware of TWatch.
is 'it possible ?
and how can restore it ?
thanks for help
Regards

ShotokuTech, 2020-09-05:
When I received my T-Watch 2020 v1 early on in production it shipped with the 'SimpleWatch" example sketch in Arduino IDE. This allows one to set WiFi and sync time. This Git repository your are on right now is the source of the code. (Xinyuan-LilyGO / TTGO_TWatch_Library). I have these videos that may be of use to you for this instance.
TTGO T-Watch 2020 Prepare the Arduino IDE
https://youtu.be/JNqu9sfF0tE
TTGO T Watch 2020 Example Sketches
https://youtu.be/7L9q-IFh2LQ

If your T-Watch 2020 v1 is shipping with the newer firmware with the weather app, multiple screens and other features, this video covers loading that. (sharandac / My-TTGO-Watch, https://github.com/sharandac/My-TTGO-Watch)
TTGO T-Watch 2020 Firmware Update Complete Success
https://youtu.be/7vOaqnmAkcg
Hope this helps

lewisxhe, 2020-09-10:
@ShotokuTech  said is right. I think your problem has been solved. If there are any problems, please reopen

closed 2020-09-10 by lewisxhe