Xinyuan-LilyGO/TTGO_TWatch_Library #79: PCF8563T CLKOUT pin connection of the T-Watch-2020

happyme531, 2020-09-23:
I want to use that signal to trigger the screen update every 1 second, but I can't find the connection to the esp32.
Not everyone can be satisfyed with delay(1000)

NorthernDIY, 2020-09-24:
I don't believe there is a connection between clkout of the PCF8563 and the ESP32.  There are available pins on the ESP32 if you feel like doing some micro soldering.  GPI34 for example is available as an input and isn't too awkward to get at.

I've done some work on mapping the pinouts, see here: [Hardware Info](https://github.com/NorthernDIY/My-TTGO-Watch/tree/experimental/Hardware%20Info)

(Note that there may be some incorrect items in this table, it is very much so a work in progress item)

lewisxhe, 2020-10-04:
Clk out is not used in the current version, and 32.768 will be added in the new version

closed 2020-10-04 by lewisxhe