Xinyuan-LilyGO/TTGO_TWatch_Library #82: i2cProcQueue(): Bus busy, reinit

LoloBee, 2020-09-29:
Hello, I've writen an app for T-Watch-2020 which works flawlessly.
When I flashed it over another T-Watch-2020 I got this error:

`[esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit`

Whenever I touch the screen.
If I keep the screen touched, after some seconds, I get this error:

```
16:26:37.454 -> [I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
16:26:37.547 -> [E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbef0c
16:26:37.547 -> [I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60027000 date=0x16042000
16:26:37.547 -> [I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffbad28
16:26:37.547 -> [I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=1
16:26:37.547 -> [I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
16:26:37.547 -> [I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
16:26:37.547 -> [I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=5
16:26:37.547 -> [I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffbadac bits=312
16:26:37.547 -> [I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffbaaf0
16:26:37.547 -> [I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffe9ab0
16:26:37.547 -> [I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
16:26:37.547 -> [I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
16:26:37.547 -> [I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=-1
16:26:37.547 -> [I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
16:26:37.547 -> [I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
16:26:37.547 -> [I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
16:26:37.547 -> [I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
```

In other T-Watch-2020 it does not happen.

LoloBee, 2020-10-01:
Just an update, I've tried the example providec by the library "Basic Unit -> TouchPad" and I got the same behaviour.

datacute, 2020-10-02:
I don't have an explanation of why it happens on one watch and not another, but I have had trouble with reading touch details to show on a web page, while concurrently polling the touch sensor.
@lewisxhe Can the Wire1 i2c "port" maybe be wrapped in an I2CBus class for the touch device so that mutex are used to restrict multiple usage of the same bus?

LoloBee, 2020-10-02:
The code works in 3 different t-watch-2020 devices, I think this may be due to a hardware malfunction.

lewisxhe, 2020-10-04:
Yes, I have encountered this situation. If it is still the same after restarting, it is usually damaged.  Please contact customer service for after-sales.
@LoloBee

lewisxhe, 2020-10-04:
For non-shared buses, there is no need to add mutual exclusion! @datacute

datacute, 2020-10-04:
@lewisxhe but it has multiple cores - I was experiencing a web page querying the state of the touch screen (only for the purposes of trying to figure out what it was doing) but that request was running in a separate task, at the same time as the polling of the touch data. That was crashing quite often until I protected the use of the Wire1 bus. I'll submit a pull request for consideration.

datacute, 2020-10-04:
For reference ... this is the error I was getting that made me think my error was related to LoloBee's:

```
[W][AsyncTCP.cpp:950] _poll(): rx timeout 4
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbfb24
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60027000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffbb02c
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=1
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=5
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffbb0b0 bits=312
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffbadb8
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3fff18ac
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=-1
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
[W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(23)=0, scl(32)=1
[I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
[I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
[I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
[W][AsyncTCP.cpp:950] _poll(): rx timeout 4
[I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
[I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
[I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
[E][esp32-hal-i2c.c:776] emptyRxFifo(): RxEmpty(4) call on TxBuffer? dq=0
Guru Meditation Error: Core  0 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 0 register dump:
PC      : 0x4013a596  PS      : 0x00060130  A0      : 0x8013ac9c  A1      : 0x3fff1100
A2      : 0x00000020  A3      : 0x3fff119c  A4      : 0x00000032  A5      : 0x3ffbfb24
A6      : 0x00004e20  A7      : 0x60027000  A8      : 0x8013a584  A9      : 0x3fff10e0
A10     : 0x00000000  A11     : 0x00000010  A12     : 0x00000000  A13     : 0x00000001
A14     : 0x00002710  A15     : 0x3ffbfb24  SAR     : 0x00000013  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x4013a596:0x3fff1100 0x4013ac99:0x3fff1140 0x400ec315:0x3fff1160 0x400ec3eb:0x3fff1190 0x400ec485:0x3fff11c0 0x400ee725:0x3fff11e0 0x400ee7dd:0x3fff1200 0x400ee7fa:0x3fff1230 0x400eb4ad:0x3fff1250 0x4011cf2d:0x3fff12a0 0x4011a665:0x3fff12e0 0x4011a733:0x3fff1320 0x4011a9b9:0x3fff1360 0x40242901:0x3fff1380 0x4024297d:0x3fff13c0 0x40243132:0x3fff13e0 0x400934f9:0x3fff1410
  #0  0x4013a596:0x3fff1100 in i2cDumpI2c at C:\Users\Stephen\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-i2c.c:318
      (inlined by) i2cProcQueue at C:\Users\Stephen\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-i2c.c:1299
  #1  0x4013ac99:0x3fff1140 in _ledcSetupTimerFreq at C:\Users\Stephen\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-ledc.c:256
  #2  0x400ec315:0x3fff1160 in TwoWire::endTransmission(bool) at C:\Users\Stephen\.platformio\packages\framework-arduinoespressif32\libraries\Wire\src/Wire.cpp:149       
  #3  0x400ec3eb:0x3fff1190 in TwoWire::requestFrom(unsigned char, unsigned char, unsigned char) at C:\Users\Stephen\.platformio\packages\framework-arduinoespressif32\libraries\Wire\src/Wire.cpp:149
  #4  0x400ec485:0x3fff11c0 in _GLOBAL__sub_D__ZN7TwoWireC2Eh at C:\Users\Stephen\.platformio\packages\framework-arduinoespressif32\libraries\Wire\src/Wire.cpp:149       
      (inlined by) _GLOBAL__sub_D__ZN7TwoWireC2Eh at C:\Users\Stephen\.platformio\packages\framework-arduinoespressif32\libraries\Wire\src/Wire.cpp:371
  #5  0x400ee725:0x3fff11e0 in FocalTech_Class::readRegister8(unsigned char) at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\drive\fx50xx/focaltech.cpp:72
  #6  0x400ee7dd:0x3fff1200 in FocalTech_Class::writeBytes(unsigned char, unsigned char*, unsigned char) at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\drive\fx50xx/focaltech.cpp:72
  #7  0x400ee7fa:0x3fff1230 in FocalTech_Class::writeBytes(unsigned char, unsigned char*, unsigned char) at .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\drive\fx50xx/focaltech.cpp:72
  #8  0x400eb4ad:0x3fff1250 in std::_Function_handler<void (AsyncWebServerRequest*), asyncwebserver_start()::{lambda(AsyncWebServerRequest*)#4}>::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*&&) at src\webserver/webserver.cpp:227
      (inlined by) _M_invoke at c:\users\stephen\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #9  0x4011cf2d:0x3fff12a0 in AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at c:\users\stephen\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h:127 (discriminator 1)
  #10 0x4011a665:0x3fff12e0 in AsyncWebServerRequest::_parseLine() at .pio\libdeps\ttgo-t-watch\ESP Async WebServer\src/WebRequest.cpp:1001
  #11 0x4011a733:0x3fff1320 in AsyncWebServerRequest::_onData(void*, unsigned int) at .pio\libdeps\ttgo-t-watch\ESP Async WebServer\src/WebRequest.cpp:1001
  #12 0x4011a9b9:0x3fff1360 in AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*) at .pio\libdeps\ttgo-t-watch\ESP Async WebServer\src/WebRequest.cpp:1001
[... cut ...]
  #14 0x4024297d:0x3fff13c0 in _tcp_recv(void*, tcp_pcb*, pbuf*, signed char) at .pio\libdeps\ttgo-t-watch\AsyncTCP\src/AsyncTCP.cpp:1020
  #15 0x40243132:0x3fff13e0 in _async_service_task(void*) at .pio\libdeps\ttgo-t-watch\AsyncTCP\src/AsyncTCP.cpp:1020
      (inlined by) _async_service_task at .pio\libdeps\ttgo-t-watch\AsyncTCP\src/AsyncTCP.cpp:194
  #16 0x400934f9:0x3fff1410 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

lewisxhe, 2020-10-04:
Are you using twatch 2020 v1?

Wire1 is provided for touch use alone, it should not be occupied, unless you use multitasking and Wire1 is used in multitasking, otherwise it cannot be occupied

@datacute

datacute, 2020-10-04:
@lewisxhe yes that's from my twatch 2020 v1 - I'm loving programming it - thank you so much!
I am trying to see whether the touch sensor can go into monitoring mode faster than 30 seconds after last touch. To do this I added a web page to show the mode it was in. That resulted in crashes due to concurrent requests for information over i2c. 
This pull request has the extra features of the touch sensor, and now also has changes to enable the use I2CBus with its mutex with a compile time define.
https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/pull/84
I wasn't able to get it to go into monitoring mode any faster than 30 seconds after the last touch, so I'm looking at changing the polling of the touch sensor to be enabled when the interrupt fires, and then checking whether the interrupt line is still low, and if not, disabling polling, and setting the touch sensor into monitor mode then.

datacute, 2020-10-04:
This is the touch reading code I'm experimenting with:

```
volatile bool DRAM_ATTR touch_irq_flag = false;
portMUX_TYPE DRAM_ATTR Touch_IRQ_Mux = portMUX_INITIALIZER_UNLOCKED;
void IRAM_ATTR touch_irq( void );
bool touched = false;

void IRAM_ATTR touch_irq( void ) {
    portENTER_CRITICAL_ISR(&Touch_IRQ_Mux);
    touch_irq_flag = true;
    portEXIT_CRITICAL_ISR(&Touch_IRQ_Mux);
}

static bool touch_read(lv_indev_drv_t * drv, lv_indev_data_t*data) {
    portENTER_CRITICAL( &Touch_IRQ_Mux );
    bool temp_touch_irq_flag = touch_irq_flag;
    touch_irq_flag = false;
    portEXIT_CRITICAL( &Touch_IRQ_Mux );
    touched |= temp_touch_irq_flag;
    if (touched) {
        data->state = touch_getXY(data->point.x, data->point.y) ?  LV_INDEV_STATE_PR : LV_INDEV_STATE_REL;
        touched=digitalRead(TOUCH_INT) == LOW;
        if (!touched) {
            TTGOClass::getWatch()->touchToMonitor();
        }
    } else {
        data->state = LV_INDEV_STATE_REL;
    }
    return( false );
}
```

lewisxhe, 2020-10-12:
I can probably understand what you mean, do you want to use interrupts to get touch status? In fact, I think it can work without adding mutual exclusion. I will find time to test it. The advantage of this is that it can save CPU resources, right! 😀

datacute, 2020-10-12:
The reason for requesting I2CBus use with its mutual exclusion was because I also had this code:

```
  asyncserver.on("/touch", HTTP_GET, [](AsyncWebServerRequest *request) {
    TTGOClass * ttgo = TTGOClass::getWatch();

    String html = (String) "<html><head><meta charset=\"utf-8\"></head><body><h3>Touch Parameters</h3>" +
                  "<b>Device Mode: </b>" + ttgo->touch->getDeviceMode() + "<br>" +
                  "<b>Interrupt Mode: </b>" + ttgo->touch->getINTMode() + "<br>" +
                  "<b>Control: </b>" + ttgo->touch->getControl() + "<br>" +
                  "<b>Power Mode: </b>" + ttgo->touch->getPowerMode() + "<br>" +
                  "<b>Monitor time: </b>" + ttgo->touch->getMonitorTime() + "<br>" +
                  "<b>Active period: </b>" + ttgo->touch->getActivePeriod() + "<br>" +
                  "<b>Monitor period: </b>" + ttgo->touch->getMonitorPeriod() + "<br>" +

                  "<br><b><u>System</u></b><br>" +
                  "<b>Uptime: </b>" + millis() / 1000 + "<br>" +
                  "</body></html>";
    request->send(200, "text/html", html);
  });
```
The constant polling to get x,y was happening at the same time as these requests for other information.

datacute, 2020-10-12:
The interrupt handler and level LOW test allows changing to monitor mode when the touch finishes, to save power.
I also had touch->setMonitorPeriod(250);
I don't know if it makes any difference though.

lewisxhe, 2020-10-13:
AsyncWebServer uses asynchronous tasks. Calling unprotected TTGOClass objects in different tasks will cause panic. If you need to access TTGOClass objects in multitasking, you must add a mutex, not on the I2C bus, because there is only A device, there will be no competition issues, please see the sample code

https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/blob/master/examples/BasicUnit/TouchMultitask/TouchMultitask.ino

lewisxhe, 2020-10-13:
> The interrupt handler and level LOW test allows changing to monitor mode when the touch finishes, to save power.
> I also had touch->setMonitorPeriod(250);
> I don't know if it makes any difference though.

This idea is good, I can add it

datacute, 2020-10-14:
Thank you for the TouchMultitask example @lewisxhe I understand.

lewisxhe, 2020-10-28:
If there is no problem, I close the question, if there are other questions, please reopen a new question

closed 2020-10-28 by lewisxhe