Xinyuan-LilyGO/TTGO_TWatch_Library #83: BLE modem light sleep support

addendumE, 2020-10-01:
According to [this post](https://github.com/espressif/esp-idf/issues/947)  achieving low power consumption with BLE connection active is possible only if the ESP32 internal RTC is clocked from an XTAL source. I don't think that Twatch2020 provides such a crystal but the CLKOUT signal coming from the PCF8563 could be connected to the 32K_XP input.
It requires just a wire and some passive components and it would drastically reduce power consumption (i.e 2mA avg keeping the BLE connection active!).
With this simple modification the Twatch2020 will turn from a toy into a real smartwatch !
Do you agree or am I  missing something ?

addendumE, 2020-10-01:
Found an answer by myself. 32K_XP pin (#12) seems to be  already used for I2C communication with touch controller. No way to exploit BLE modem light sleep even with manual wiring the missed connection

lewisxhe, 2020-10-04:
Yes, 33,32 has been used, 32.768 cannot be used.  A 32.768 crystal oscillator will be added in the new version, but not 2020V1

closed 2020-10-04 by lewisxhe

addendumE, 2020-10-04:
This is a great news to me. When do you think that new version Will be
released?

timstableford, 2022-04-11:
@lewisxhe did this get added in newer revisions of the watch?