Xinyuan-LilyGO/TTGO_TWatch_Library #86: use FFat error!

justeene, 2020-10-10:
FFat.format();
用FFat文件系统会报错：
[E][FFat.cpp:35] check_ffat_partition(): No FAT partition found with label ffat
[W][FFat.cpp:94] format(): No partition!

lewisxhe, 2020-10-12:
你应该是没有选择分区表 ，请看下图 。

[image]

justeene, 2020-10-12:
> 你应该是没有选择分区表 ，请看下图 。
> 
> [image]

如果是vscode+platformio怎么配置分区表呢？
我的platformio.ini配置如下：

`
[env:MyWatch]
platform = espressif32
board = ttgo-t-watch
framework = arduino
lib_archive = false
board_build.f_flash = 80000000L
monitor_speed = 115200
monitor_filters =
    default
    esp32_exception_decoder
build_flags =
    -DCORE_DEBUG_LEVEL=3
    -mfix-esp32-psram-cache-issue
src_filter =
    +<*>
`

lewisxhe, 2020-10-12:
```
board_build.partitions = default_ffat.csv
```

justeene, 2020-10-13:
问题解决。
我用的是：
board_build.partitions = app3M_fat9M_16MB.csv
[image]

closed 2020-10-13 by justeene