Xinyuan-LilyGO/TTGO_TWatch_Library #87: No LoRa performing on TTGO T-watch 2019 with the S7XG LORA/GPS board

xavierflorensa, 2020-10-11:
I have a TTGO ESP32 watch with the S7XG LORA/GPS board. I'm not able to get any performance with LoRa. And there is no explanation on the following sketch. 

https://raw.githubusercontent.com/Xinyuan-LilyGO/TTGO_TWatch_Library/master/examples/Shield/S7xG_LoRa/S7xG_LoRa.ino

Any link to find out more info on how to take profit of this LoRa Board? Since the board is intended to be used as a human tracker, I would like to know, Is it compatible with a LoRaWAN Gateway? Where do you adjust the frequency to 868Mhz EU?

lewisxhe, 2020-10-12:
S7xG in Twatch lib is just a demonstration. For more functions, please see here. This library is not compatible with the new Twatch lib, but the main functions are still the same. You can refer to here.
The frequency can only be modified by burning the fixed Acsip firmware, and cannot be modified in the software

https://github.com/lewisxhe/Acsip-S7xG-Library

xavierflorensa, 2020-10-18:
Thank you very much, I really appreciate. I will try

xavierflorensa, 2020-10-18:
Dear Lewis, I'm compiling your library 
https://github.com/lewisxhe/Acsip-S7xG-Library/tree/master/examples/lorawan_abp
with Arduino IDE and the compiler gets stuck on following line 19
TFT_eSPI *tft;
With following error:
error: 'TFT_eSPI' does not name a type
Would you be so kind to suggest how can I solve this error? Thank you very much

lewisxhe, 2020-10-19:
It was written by the old version of the Twatch library, I have updated it, please download and test again

xavierflorensa, 2020-10-19:
Great! thank you very much, I'm downloading the last version of 
https://github.com/lewisxhe/Acsip-S7xG-Library
I will test it and give feedback :-)

xavierflorensa, 2020-10-19:
Dear Lewis, Yes, now it compiles and uploads to T-Watch219 with LoRa/GPS Module, but you see on the screen, the blue colored ACsip Logo over a black color but also an error message:
Initialization 'S76/S78G' failed
corresponding to code line

hwSerial = new HardwareSerial(1);
    hwSerial->begin(115200, SERIAL_8N1, GPS_RX, GPS_TX);
    if (!s76g.begin(*hwSerial)) {
        tft->setTextColor(TFT_RED);
        tft->println("Initialization 'S76/78G' failed");
        while (1);
    }

Do you think the LoRa/GPS Module has no power?
How can I check it?
Thanks and Best regards

xavierflorensa, 2020-10-19:
[image]

xavierflorensa, 2020-10-19:
Dear Lewis, I have opened the watch to see if the S67G chip is powered but I see that there is another chip, see pictures. So which library I'm supposed to use for LoRaWAN applications? Thank you very much
[image]
[image]

lewisxhe, 2020-10-20:
You are not the bottom plate of S76G, but the bottom plate of Lora (sx1276), you should use this
https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/master/examples/Shield/LoRa_SX127x

xavierflorensa, 2020-10-20:
Thank you very much Lewis, but this one is not LoRaWAN, it is just LoRa base
https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/master/examples/Shield/LoRa_SX127x
I will try with other TTGO libraries for TTN like the TTO T-Beam

lewisxhe, 2020-10-28:
Sorry, I see that there is a GPS antenna installed on your board. It is wrong. Please remove it. Although it will not affect it, I have to explain to you that there are many examples of TTN on github. I will close the issue, if there are other issues, please open a new issue

closed 2020-10-28 by lewisxhe