Xinyuan-LilyGO/TTGO_TWatch_Library #88: PN532 example not working

fsalomon, 2020-10-13:
Hi, I have the `LILYGO_WATCH_2019_WITH_TOUCH` with `LILYGO_WATCH_HAS_NFC`. The factory installed app shows "Waiting for an ISO14443A Card ..." forever i.e. not recognizing my tag. It is a Mifare Classic Tag. The tag works fine with https://play.google.com/store/apps/details?id=de.syss.MifareClassicTool

https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/blob/master/examples/Shield/NFC_Lock/NFC_Lock.ino with `#define PN532DEBUG` gives me this output:

```
23:12:57.095 -> Hello!
23:13:00.447 -> 
23:13:00.447 -> Sending: IRQ received
23:13:00.447 ->  0x0 0x0 0xFF 0x0 0xFF 0x0
23:13:00.481 ->  0x0 0x0 0xFF 0x6 0xFA 0xD5 0x3 0x32 0x1 0x6 0x7 0xE8
23:13:00.481 -> Found chip PN532
23:13:00.481 -> Firmware ver. 1.6
23:13:00.481 -> 
23:13:00.481 -> Sending: IRQ received
23:13:00.514 ->  0x0 0x0 0xFF 0x0 0xFF 0x0
23:13:00.514 ->  0x0 0x0 0xFF 0x2 0xFE 0xD5 0x15 0x16
23:13:00.514 -> Waiting for an ISO14443A Card ...
23:13:00.514 -> 
23:13:00.514 -> Sending: IRQ received
23:13:00.547 ->  0x0 0x0 0xFF 0x0 0xFF 0x0
23:13:00.547 -> Waiting for IRQ (indicates card presence)
```

Is this just faulty hardware or am I missing something?

fsalomon, 2020-10-14:
Was a hardware fault. Fixed with a soldering iron.

closed 2020-10-14 by fsalomon

piyapoonsawat, 2020-12-04:
Do you have any solution? I am planning to buy one.

fsalomon, 2020-12-04:
Yes mine is working

piyapoonsawat, 2020-12-04:
Thank you, fsalomon.