Xinyuan-LilyGO/TTGO_TWatch_Library #93: BLE beacon for presence detection

anarchking, 2020-11-01:
I  think it would be very useful to have an app that you click a button and it turns on the ble Beacon Broadcasting so other devices can be scanning in all rooms so you're so you're smart home can recognize what room you're in.

anarchking, 2020-11-01:
Example: https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/examples/BLE_iBeacon/BLE_iBeacon.ino

lewisxhe, 2020-11-30:
What do you want to express?

careyer, 2020-11-30:
Hi Lewis,

 

I think the suggestion was that there should be an option to continously transmit a BLE beacon from the Watch
so that sensors placed arround you in the environment (like in your smarthome) can track in which room you
are and trigger actions accordingly. Basically "Presence detection" via means of BLE. Many Bluetooth gadgets
lilke sport trackers and alike do this. This would really be a nice addition.

Best regards
careyer
 

Gesendet: Montag, 30. November 2020 um 05:01 Uhr

anarchking, 2020-11-30:
I looked deep into this esp32 ble and it seems that the esp32 has trouble being a true Bluetooth "LOW ENERGY" with out adding a crystal to it. Its would be nice to get it as low as possible by setting the pace it beacons "broadcasts or ppm i think" and tx power (distance or rssi signal). Advertising sensor values

lewisxhe, 2020-12-29:
@careyer  As @anarchking  says, esp32's Bluetooth is not "low energy" in the true sense. It has no meaning at all as a battery-powered beacon, unless you have been plugged in USB.

If you need sample code, you can refer here

https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE/examples/BLE_iBeacon