Xinyuan-LilyGO/TTGO_TWatch_Library #94: CRC error when fetching libraries on platformio

raomin, 2020-11-01:
`Library Manager: Installing TTGO TWatch Library @ =1.3.0
Unpacking...
Library Manager: Warning! Package Mirror: CRC check failed 0xed05a807 != 0x23a65ce7
Library Manager: Looking for another mirror...
Unpacking...
Library Manager: Warning! Package Mirror: CRC check failed 0xed05a807 != 0x23a65ce7
Library Manager: Looking for another mirror...
Unpacking...
Library Manager: Warning! Package Mirror: CRC check failed 0xed05a807 != 0x23a65ce7
Library Manager: Looking for another mirror...
PackageException: Could not install package 'TTGO TWatch Library @ =1.3.0' for 'windows_amd64' system:
  File "C:\Users\bourgur\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:`

Please also have a look here, https://github.com/platformio/platformio-home/issues/1605

lewisxhe, 2020-11-05:
In my case, I have no problem, I think it is a network problem, you can clone this repository directly into the lib directory

closed 2020-11-05 by lewisxhe

raomin, 2020-11-06:
It was a temporary issue in the PIO repository.: https://github.com/platformio/platformio-home/issues/1605#issuecomment-721097983 
Fixed now.