Xinyuan-LilyGO/TTGO_TWatch_Library #95: Bug? TTGOClass::touchToSleep() does not disable the touch driver

SpectralCascade, 2020-11-03:
I expected this mode to disable the touch input driver, but I'm still getting touch interrupts and when I query the touch driver I get valid touch data as though it is not asleep.

lewisxhe, 2020-11-24:
Sorry, I have been busy lately. Are you using twatch-v1 or twatcg-2109? I will test it

SpectralCascade, 2020-12-09:
Hiya, I am using ttgo twatch-2020 v1

Netspore, 2021-01-27:
I also have a ttgo twatch-2020 V1 and would be interested in knowing if there's a way to put the touch sensor into low power mode.  It doesn't appear that the WAKE signal is implemented (or is it)?

lewisxhe, 2021-01-28:
I have added a test example for the three modes of touch, and the test results are very good. Please note that only TWATCH 2020-V2/2020-V3 has touch wake-up, none of the others support  @Netspore

closed 2021-01-28 by lewisxhe