Xinyuan-LilyGO/TTGO_TWatch_Library #97: Osmand Howto

Neuroplant, 2020-11-12:
Hello,
can you give an short instruction how to connect Osmand to the watch?

thx
Neuroplant

Neuroplant, 2020-11-18:
found answer
[https://github.com/sharandac/My-TTGO-Watch#bluetooth](https://github.com/sharandac/My-TTGO-Watch#bluetooth)

closed 2020-11-18 by Neuroplant

hb9eue, 2021-05-30:
Linkd does not seem to point to the right direction anymore. I have the same question :-)

hb9eue, 2021-05-30:
Ok, answering myself after some more google research and testing.
This is how I got this to work
You need gadgetbridge and osmand+
osmand+ send 'notifications' about directions. Those are passed to the watch by gadgetbridge.
I had to set 'send notification when display on' on gadgetbride.
And you need to move. The first 'direction' from within osmand+, right after clicking on 'navigate' was not being sent, only 'updates'.