Xinyuan-LilyGO/TTGO_TWatch_Library #98: Feature Request: ESP3D

ETE-Design, 2020-11-16:
I'm really not into programming... But will it be possible to make it work with ESP3D so it can send status and and let you change settings direct from the watch?

lewisxhe, 2020-11-24:
What is ESP3D?

Fusseldieb, 2020-11-24:
@lewisxhe Maybe this: https://github.com/luc-github/ESP3D

ETE-Design, 2020-11-24:
@lewisxhe as @Fusseldieb  Link to... But the GUI is this https://github.com/luc-github/ESP3D-WEBUI/tree/3.0
World be great to be possible to control / see status from the watch...

lewisxhe, 2020-11-30:
This is the firmware of the 3D printer, do you want the watch to control it?

ETE-Design, 2020-11-30:
@lewisxhe No it is not, it is a GUI that make it possible to to control the 3D Printer from anywhere as Long you have a browser... The firmware for 3D Printer is Calles Marlin so not the same...

ETE-Design, 2020-12-02:
@lewisxhe The main goal would be to pull the info the WebUI - [https://github.com/luc-github/ESP3D-WEBUI/tree/3.0](url) gets from the 3D Printer, but I'm really not good to programming so don't know how to get that info from a "Browser" to a watch version?

lewisxhe, 2020-12-29:
Sorry, I don’t understand this, I can’t help you.