Xinyuan-LilyGO/TTGO_TWatch_Library #101: Many crashes of TWatch during physical activities

destroyedlolo, 2020-11-30:
Hello,

I'm suffering of TWatch crashes using SampleWatch or derivative during physical activities.
It's seems related to too many IRQ in short period of time : it seems the g_event_queue_handle is not large enough and FreeRTOS not protected against overflow.

I increased this queue to 60 and my watch (with my own code) was stable for 2 days even if I did physical activities (mountain hiking).

Can you please upgrade FreeRTOS in the library ?
You can also upgrade Lvgl to the latest version : I tested and it's working pretty well w/ TTGO library :)

Thanks

Laurent

lewisxhe, 2020-12-01:
Freertos is internal to esp32, not TTGO Lib. I have upgraded lvgl to the latest version 7.7.2.

destroyedlolo, 2020-12-01:
Ok, so I'll see with https://github.com/espressif/arduino-esp32
Thanks a lot

Laurent

closed 2020-12-29 by lewisxhe