Xinyuan-LilyGO/TTGO_TWatch_Library #106: Touch interrupt broken

SpectralCascade, 2021-01-09:
For some reason the touch interrupt stopped working when I updated for the T-Watch 2020 V1. It seems that the touch controller is never initialised properly. I fixed it by reverting commit 16de81b53f6a46be9605ca56da65a3d96ce09ac8
which seems indicative of an issue with the definition of LILYGO_TOUCHSCREEN_CALLBACK_METHOD

SpectralCascade, 2021-01-09:
Also related I think, focaltech.cpp FocalTech_Class::getTouched() always returns 0.

lewisxhe, 2021-01-12:
I tested well here, and there is no problem you said

SpectralCascade, 2021-01-13:
> I tested well here, and there is no problem you said

Hm, that is strange. I cleaned the library object files and rebuilt it from scratch and it now works, must have been an issue with the build linking some old code perhaps. Closing this as a non-issue

closed 2021-01-13 by SpectralCascade