Xinyuan-LilyGO/TTGO_TWatch_Library #107: How to flash game rom for NES example ?

superuser789, 2021-01-10:
I am using TTGO T-Watch 2020.  I have successfully flashed NES example using Arduino 1.8.13 in Windows 10. The watch boots and goes blank with backlight on. 
In [here](https://github.com/espressif/esp32-nesemu), it is mentioned that I need to flash my nes rom to the address 0x00100000.
I tried :
`esptool --chip esp32 --port "COM3" --baud 921600 write_flash -fs 16MB 0x00100000  gamrom.nes`

The watch powers off and does not turn on using power button.

lewisxhe, 2021-01-12:
Please upload to spiffs with esp32fs , If you have not installed it, please see here
https://github.com/me-no-dev/arduino-esp32fs-plugin
[image]

superuser789, 2021-01-12:
> Please upload to spiffs with esp32fs , If you have not installed it, please see here
> https://github.com/me-no-dev/arduino-esp32fs-plugin
> [image]

Thanks. It worked.
Is loading multiple roms supported ? or I need to upload one game rom at a time ?

lewisxhe, 2021-01-13:
This example does not support multiple games, but it can be easily changed to multiple games, you only need to add a selection interface.

closed 2021-01-25 by lewisxhe