Xinyuan-LilyGO/TTGO_TWatch_Library #108: ESP-Now on T_Watch 2020

middelbh, 2021-01-18:
Hi, I am working on ESP-Now on my new T-Watch 2020. I cannot send or recieve data to and from an ESP32 devkit. Is there any experience with ESP-NOW on the T-Watch 2020?

lewisxhe, 2021-01-19:
Run [ESPNow](https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/ESPNow) directly, you need at least two devices, and burn the `Master` separately and `Slave`

middelbh, 2021-01-19:
Lewis, thanks for youranswer. These scetches work if I use channel 1; if I use channel 1 on master (ESP32 devkit) and channel 3 on slave (twatch 2020) they do not transmit data, but master says succesful and slave does nnot print. I would like to use twatch as STA, not as AP. I nowhave the feeling that theyare not on the same channel... will do some more tests tomorrow, but I cannot dind channelsetting for STA mode...

middelbh, 2021-01-21:
Did a lot of tests now. T-watch works with AP mode, but not in STA mode, like I said earlier, in STA mode on my t-watch sending message from my devkit responds that message is sent successfully, but watch does not respond. Please help!

lewisxhe, 2021-01-21:
Have you tried the official example?  No need to modify it, just run it directly.  This obviously has nothing to do with the hardware, I think the problem may be in the software.

middelbh, 2021-01-21:
I used the programs as you proposed, works without problems when in AP mode. But after changing the t-watch to STA mode like in the approach of randomnerdstutorials and the devkit to its counterpart, the devkit reacts with successful sent the message, but the watch seems to not receive data. I thoroughly checked MAC addres of the t-watch!

lewisxhe, 2021-01-28:
I use two T-Watch2020 V1, they burn Base/Master (STA Mode), Base/Slave (AP Mode), I did not modify the code, they run well

middelbh, 2021-01-28:
Lewis, thanks! I will test again. Maybe I need to completely clear SPIFSS of the watch, I will try!

middelbh, 2021-01-30:
Lewis,

Problem solved! I have completely reinstalled Arduino IDE and all libraries and now ESP-NOW also works in STA mode. Thanks a lot for your help!

closed 2021-01-30 by middelbh