Xinyuan-LilyGO/TTGO_TWatch_Library #115: s7xg gps not working

asavvin, 2021-03-08:
Trying to use library example s7xg. Keep getting the connecting.... even outside.
Is there any  way to check hardware, also is it possible to use termite interface I tried and seems to me I can't send any commands.

jdevicente10, 2021-03-10:
I have the same problem when loading any program to the TWatch v.2019 with the s76g module, however, for another TWatch v.2020 I am able to program it correctly. I have tried to press and hold the boot button while loading the program, but I keep getting the same error message. Can anyone help us?

[image]

lewisxhe, 2021-03-27:
You can try to use https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/master/examples/Shield/S7xG_Debug for debugging. The built-in functions of TWatch lib are not updated in time. For more information, please check here https://github .com/lewisxhe/Acsip-S7xG-Library

lewisxhe, 2021-03-27:
@jdevicente10 If you can’t upload when it’s turned on, the automatic download circuit may be damaged.

asavvin, 2021-04-06:
> You can try to use https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/master/examples/Shield/S7xG_Debug for debugging. The built-in functions of TWatch lib are not updated in time. For more information, please check here https://github .com/lewisxhe/Acsip-S7xG-Library

Hi Thank you for the code, however I don't see code in a library anymore, somebody remove the code. Could you please update Shield/S7xG_Debug

lewisxhe, 2021-04-09:
https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/master/examples/Shield/S7XG/S7xG_Debug