Xinyuan-LilyGO/TTGO_TWatch_Library #120: LILYGO WATCH 2020 V1 wake from light sleep touch issue ?

YordanYanakiev, 2021-03-28:
I am trying to use touch screen of V1 of the watches ( LILYGO_WATCH_2020_V1 ) but when it goes to light sleep and wake up - the touch does not works.
Can You please add minimalistic example of using light sleep with touchpad with LILYGO_WATCH_2020_V1 ?

closed 2021-04-08 by YordanYanakiev

destroyedlolo, 2021-04-09:
Hello,

Did you solved your problem ? 
I don't have any issue when wake up from light sleep but the touch screen controler is trashed when wakeup from deep sleep.

Bue

Laurent

YordanYanakiev, 2021-04-09:
well..  the issue happens once per let say 100-500 cycles.
sometimes rare, some time not so. but it happens.

destroyedlolo, 2021-04-09:
Hum,

My concern is only with deep sleep : I'll have to rework Lewis' code to split C code initialisation and hardware one.
- as deep sleep is like a reboot, the C initialisation need to be done
- but the hardware is already initialised (no poweroff, no reset) and it doesn't like to get calibrated twice.
But I have more important stuffs to work on before :)

Back to your problem with light sleep : I'm not suffering of this problem at all (and I'm wearing my watch every days for months now) : so my [own code](https://github.com/destroyedlolo/DomoWatch) is solid as a rock on this domain.
I suspect something wrong in your own ... OR it may be several hardware release for the v1 watch but only Lilygo can reply on this aspect.

YordanYanakiev, 2021-04-09:
This sounds very promising actually. Probably then the issue is with me. I will check Your code to get the correct sequence then.
O yeah. Btw - how do You handle interupts from the button while for instance the watch is working on something - like downloading data or calculating or drawing stuff ? Are You threads based ?

destroyedlolo, 2021-04-09:
Yes, it's threads based but graphics are handled only in the "main" thread (because it seems LVGL is not really multi threading compliant).
So events are handled in sub thread and communicating with the main one using signal if needed. It's the case also for the PEK button : it's up to the main thread to decide if it can go to sleep or if something is on way.

By the way, networking is not already coded for the watch as I'm currently working on backends that will provide data to the watch (close to be finished).