Xinyuan-LilyGO/TTGO_TWatch_Library #122: TTGO T-Watch 2020 V1 WiFi issue.

YordanYanakiev, 2021-04-08:
I am facing a strange problem.
I am setuping a WiFiMulti with 2 APs on the list.
Then I connect with HTTPClient to a web server and get a text data.
Then put the board into sleep mode.
Wake up the board in 10 seconds.
And repeat this forever.
This works fine on "ESP32 Devkit v1" module, but on "T-Watch 2020 V1" randomly disconnects from the WiFi or the HTTPClient stop working normally.

Please check this issue.

YordanYanakiev, 2021-04-08:
https://github.com/espressif/arduino-esp32/issues/5020

lewisxhe, 2021-04-09:
You can try the WiFiMulti example first to see if it is running normally. If it is running normally, hardware problems can be ruled out. If it is running abnormally, then it is definitely a hardware problem.

YordanYanakiev, 2021-04-09:
That's basic thing, and it is not applicable.
We are speaking about issue which happens randomly over relation HTTPClient + Wifimulti + Light Sleeping.

lewisxhe, 2021-04-09:
Then if the basic example runs normally, then the hardware problem can be ruled out.

As for the software issue, I can’t help you deal with it

YordanYanakiev, 2021-04-09:
This example is not related to the problem.
As I have said - it's a complex problem related to the HTTPClient + Wifimulti while going into and waking from Light Sleeping.

fraschizzato, 2021-08-31:
I'm facing a similar problem. The watch works correctly as access point and with the wireless scan example, but I haven't found any way to get it work as station. Tried wifi client example, multi and the easy wifi. Tried on different ap. The same code works on other esp32 devices.

KLip9, 2021-12-09:
I have the same problem. In some cases the wifi connection on my TTGO T-Watch 2020 V1 is not reliable. This can be seen in the followed screenshot from wireshark. The IP-address of the watch is 192.168.10.26. The server is on 192.168.10.1. You can see that there are some TCP Retransmissions. That's in my case the reason for the connection bug. The watch doesn't receive the response. In some cases the request is successful. 

The programm also uses HTTPClient + Wifimulti.

[image]

YordanYanakiev, 2021-12-09:
Check this as well fellow KLip9. Your work could be useful ! 
https://github.com/espressif/arduino-esp32/issues/2501

KLip9, 2021-12-15:
Thanks for your response @YordanYanakiev.
In your posted issue I saw some workarounds. I tried a lot and finally I write a script to get a reliable WIFI connection. Unfortunately I don't use the WIFIMulti libaray anymore and I have to configure a static IP-address. 

This is my script to get a reliable connection:

  ```
  #define LILYGO_WATCH_2020_V1
  #define LILYGO_WATCH_LVGL
  
  #include <LilyGoWatch.h>
  #include <WiFi.h>
  #include <HTTPClient.h>
  
  void connectToWifi(void)
  {
      if (WiFi.status() == WL_CONNECTED) {return;}
      WiFi.config(IPAddress(192,168,2,3), IPAddress(192,168,2,1), IPAddress(255,255,255,0));
      WiFi.begin("SSID", "PW");
      WiFi.persistent(false);
      WiFi.setTxPower(WIFI_POWER_2dBm);  //maybe unnecessary??
      while (WiFi.status() == WL_DISCONNECTED) {
          Serial.println("Waiting for WiFi to leave WL_DISCONNECTED...");
          delay(500);
      }
      delay(500);
  }
  
  void setup(void)
  {
      Serial.begin(115200);
      WiFi.mode(WIFI_STA);
  }
  
  void loop(void)
  {
      bool isConnected = false;
      Serial.println("Check connection...");
      while (!isConnected) {
          connectToWifi();
          
          Serial.println("Try to ping...");
          HTTPClient client;
          client.begin("SOME URL");
          const auto httpCode = client.GET();
          client.end();
  
          isConnected = (httpCode == HTTP_CODE_OK);
          if (isConnected) {Serial.println("Ping OK");}
          else {Serial.println("Ping failed");}
      }
      Serial.println("Watch is connected");
      Serial.println(WiFi.localIP());
  
      // Do some other requests ..
  
      delay(1000);
  } 
```