Xinyuan-LilyGO/TTGO_TWatch_Library #131: I2S Sound with MAX98357A does not work for T-Watch-2020-V3

anton-mir, 2021-05-17:
Hi,
Despite proper connections ESP32<->MAX98357A->Buzzer I still cannot hear any sound from T-Watch-2020-V3.
I try ESP8266Audio library with Arduino as well with no luck.
Could you please approve that T-Watch-2020-V3's sound should work and if so please tell me how I can test it for sure.
Thanks in advance

Ruda56, 2021-05-22:
Hi,
I had same problem with T-watch 2020 v1, because pin 33 and pin 25 of ESP32 was shorted from factory.
You can try set pin 33 high and read value of pin 25. If value is also high then pins are shorted. Also try the same procedure on pins 25 and 26. 
I used this modified blink sketch to verify pins connection. If serial monitor shows alternating 0 and 1, its shorted. Otherwise it should be 0.

```
int high_pin =33;
int read_pin=25;
// he setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin high_pinN as an output.
    Serial.begin(115200);
    Serial.println ("--------------------Starting---------------------------");

  pinMode(high_pin, OUTPUT);
  pinMode(read_pin,INPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(high_pin, HIGH);   // turn the LED on (HIGH is the voltage level)
  Serial.println (digitalRead(read_pin));
  delay(1000);                       // wait for a second
  digitalWrite(high_pin, LOW);    // turn the LED off by making the voltage LOW
    Serial.println (digitalRead(read_pin));
  delay(1000);                       // wait for a second
}
```

Fortunally I was able to repair mine with fine tip soldering iron, few drops of flux and magifier glass.
Good luck.

siegmar, 2021-07-13:
Hi, maybe the solution  some Pinout has changed 
AXP202 LDO4  is now the power for the Audio Module not LD03 
I can not test ist, because I have no V3 watch in my hand 
cheers
sigges