Xinyuan-LilyGO/TTGO_TWatch_Library #133: Infrared LED not working on TWatch 2020 v3?

peteh, 2021-06-06:
I was trying to run some examples for IR control. However it seems that the LED is not working or I'm doing something wrong. 
This is the most basic code I could come up with to test: 
```
uint8_t m_pin;
    void setup()
    {
       uint8_t pin TWATCH_2020_IR_PIN; // resolves to pin 13
       m_pin = 13;
       m_on = false;
       pinMode(m_pin, OUTPUT);
       digitalWrite(m_pin, m_on);
    }

    void loop()
    {
        m_on = !m_on;
        digitalWrite(m_pin, m_on);
        delay(2000);
    }
```

This code should make the led turn on and off every 2 seconds, so I should see it in my smartphone camera (This works for other infrared leds that I'm using). 

Any ideas what I could do to further analyze if the led is broken or if I'm doing something wrong?

m00nbyt3, 2021-11-06:
Same here, if I set this #define LILYGO_WATCH_2020_V1  the IR transmitter will work, but as my watch is V3 other functions arent working correctly...

m00nbyt3, 2021-11-06:
Ive just found a solution, in the V3 theTFT BackLight is in theory in pin 15. Thats what is causing somehow mess. For solving the problem, Ive just edit the file Arduino\libraries\TTGO_TWatch_Library-master\src\board\twatch2020_v3.h, and  modified the line 
"#define TWATCH_TFT_BL               (GPIO_NUM_15)"
to 
"#define TWATCH_TFT_BL               (GPIO_NUM_12)"
I thought that will break the backlight, but it is rearely also working, and now the IR Led also works :))

peteh, 2022-03-07:
Sorry, I forgot to actually reply here. The code was good but the led was indeed broken. I ordered a second watch after customer support was sending me back and forth for trying stuff. Second one worked and I asked them to replace the first one.

closed 2022-03-07 by peteh