Xinyuan-LilyGO/TTGO_TWatch_Library #135: Unable to upload sketch

nmanning556, 2021-06-25:
Just got a T-Watch 2020 and I keep getting a time out waiting for packet header error. I've had this before on some other dev boards that just need to have the boot button held while connecting, but the one button on the t watch doesn't seem to do this. Work around? Solutions?

szmol96, 2021-07-22:
Try pressing the reset button about a second after the last dot has appeared in the "dots and lines" pattern. It needs good timing, but works every time for me.

YordanYanakiev, 2021-10-17:
Same here.
ESP32 2.0.0 framework on Visual Studio 2019 Pro + VisualMicro = can not upload the sketch.

FXVT, 2021-10-20:
Bonjour Everybody,
Arduino IDE. v1.8.13, Card manager V1.6 or 2.00
I have the same error even when pressing the reset button
_esptool.py v3.0-dev
Serial port COM8
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header_

Somtimes it works well,
Sometimes it works twice
Sometimes I have to reboot my PC to free the USB Ports
Sometimes the error message is different ...

How can we make it safely ?