Xinyuan-LilyGO/TTGO_TWatch_Library #148: Issues with touch calibration

RadTekWeb, 2021-12-16:
Hello, my T watch 2020 v2 seems to need touchscreen calibration, for example to activate the corner buttons I always need to press to their right or left, but if I click right on them nothing happens except a vibration. 

Is there a way to calibrate it or should I return my watch?

mike-ady, 2022-01-05:
I am no expert, but a quick search through the TWatch library source code did not reveal any obvious calibration routines.  The library appears to support capacitive touch sensors made by FocalTech.  I viewed a couple of the pertinent FocalTech datasheets, and the sensors appear to be self-calibrating.  

Sometimes it's hard to tell what is going on on such a small display.  You could try downloading Dan Geiger's TWatch framework.  It includes a touch app, that displays the touch location.  I find that if I lightly touch the display and drag my finger tip slowly to one of the edges, I can get within a few counts of the limits, (0 or 239).  You can find Dan's code on Instructables, here:  

https://www.instructables.com/Lilygo-T-Watch-2020-Arduino-Framework/

Be warned, that Dan's touch app times out and returns to the main time display if you don't touch the display within 5 seconds.

Gun-neR, 2022-05-14:
I just received my T-Watch 2020 V2... and can barely use it at all.  The calibration is so far out that I am basically left with tapping and swiping randomly in order to get anything to happen.  Not too impressed so far.

I have been searching on how to calibrate this thing... no idea how to load any Apps yet... not even sure I can if it requires touch at any point :P

RichFind, 2022-10-11:
I have a new V2 and a new V3, the touch screens are terrible way out of calibration.