Xinyuan-LilyGO/TTGO_TWatch_Library #153: IRremote sketch

NDC-SC, 2022-01-16:
The IRremote sketch in Examples > TTGO TWatch Library > LVGL > IRremote no longer complies,

given the error of "no matching function for call to 'IRsend::IRsend(int)'" for line "IRsend irsend(TWATCH_2020_IR_PIN);" 

Is there an update of the sketch ?

Thanks

Khalinda, 2022-01-20:
I believe that Lewis He is the maintainer of this repository.  He appears to be on hiatus, perhaps permanently.  So you/we are on our own for the time being.

I have found the IRremote library to be impossible to work with in the past.  At least some of the example sketches provided with the latest version of the IRremote library build and appear to work with the TWatch core/library.  (Disclosure:  I have a TWatch 2000 V1 and I am running the ESP32 Arduino Core v1.0.6.)  

I built and loaded the IRremote SendDemo example.  IR_SEND_PIN is defined as pin 4 in the code.  That is the motor pin on the TWatch V1, so at least I know the sketch is doing something.

In the absence of anybody else's help, your best bet may be to merge the IR code from one of the working IRremote library examples with the LVGL graphics in the TWatch IRremote code.

NDC-SC, 2022-01-21:
Thanks for your comments.
Used the libraries IRremoteESP8266.h and IRsend.h to transmit IR messages with the instruction irsend.sendRaw
For the TWatch V2, the TWATCH_2020_IR_PIN is GPIO 2.